hyperkineticnerd/OpenCorsairLink #137: H110i Fan 0 data also shown as Pump Data

rigred, 2018-09-23:
**Describe the bug**
On the H110i I have the Pump data is not displayed, instead the First Fan data is repeated as the Pump data. 
No pump state changes can be seen. But it seems that:
* Pump speeds can be set. 
* Led colors can be changed. 
* Fan speeds can be changed.

**To Reproduce**
1. Run `OpenCorsairLink --device 0`
2. View the Output
3. Confirm it is true by disconnecting Fan 0
4. Notice that the Pump speed information also falls to 0

**Expected behavior**
The Pump speed data is shown and not Fan 0 data.

**Screenshots**
[image]

**System:**
 - Arch Linux 
 - Kernel 4.18.9
 - Both OpenCorsairLink master & testing build

**Additional context**
`--debug --dump`
```text
No (device_id 0x42)
Corsair product detected. Checking if device is H110i... 
---- Packet dump: -----------------------------
8a 07 42 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
Dev=0, CorsairLink Device Found: H110i!
Checking USB device 7 (1d6b:0002)...

```

rigred, 2018-09-23:
If I want to correct this, code you give me pointer at where in the code to check?

rigred, 2018-09-23:
I've seen a few other (incomplete) issues by others that confirm the issue is correct.

https://github.com/audiohacked/OpenCorsairLink/issues/102
https://github.com/audiohacked/OpenCorsairLink/issues/136

Otherwise Related:
https://github.com/audiohacked/OpenCorsairLink/issues/65

hoanguc3m, 2018-09-23:
> If I want to correct this in the code, could you give me pointer at where to check?

My solution: I install virtual box with windows and install Corsair link. Then, use cooler as usb and change the configure.
If that could help. Bests.

audiohacked, 2018-09-23:
@rigred Check that the pump index is correct in `device.c`
Also insure the logic for the water cooler is correct in `logic/settings/hydro_coolit.c`

audiohacked, 2018-09-23:
@rigred I just made a minor fix in `device.c` for pump indexes for a few devices.

rigred, 2018-09-23:
I've tested the code and it doesn't change anything functionally. 
I know you incremented the `pump_index` values for coolit devices by 1.
But everything is still reporting the same values.

Note how Pump data is showing the same values as Fan 0.
```
sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H110i!

Vendor: Corsair
Product: H110i
Firmware: 2.0.0
Temperature 0: 29.12 C
Fan 0:  Quiet Mode (4PIN)
        Current/Max Speed 1050/2061 RPM
Fan 1:  Quiet Mode (4PIN)
        Current/Max Speed 1031/2089 RPM
Pump:   Mode 0x89
        Current/Max Speed 1050/2061 RPM
```
Fan 1 is actually reporting correct information, Fan 0 is also correct.

I haven't yet taken the time to fully immerse myself in the codebase ,but I have looked in the places you told me and there it looks ok. 
I suspect that somewhere the pump data is being overwritten with Fan data or a pointer is wrong.
I will continue to investigate.

audiohacked, 2018-09-24:
@rigred As a note, the pump functions are in `protocol/coolit/pump.c`.

NoMod-Programming, 2018-12-27:
I've submitted a fix for this in #144. Note that I'm not sure if the pump data is correct for other models, only that it works on the H110i. If someone using another pump could test it out, that'd be great!

audiohacked, 2018-12-29:
Fixed!

closed 2018-12-29 by audiohacked