hyperkineticnerd/OpenCorsairLink #138: H100i GT V2 no pump control

KevinAGI, 2018-09-26:
When I try to control the pump I get Unsupported Generic Pump Function.

Dev=0, CorsairLink Device Found: H100i GT V2!

Vendor: Corsair
Product: H100i GT V2
Firmware: 2.8.0.0
Temperature 0: 33.60 C
Temperature 1: 33.60 C
Temperature 2: 33.60 C
Fan 0:	Mode 0xE5
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
Pump:	Mode 0x00
	Current/Max Speed 3150/0 RPM

My Fan speed is zero, because they are being controlled by the motherboard.

Mac OS X 10.12.5
Running the master release

netinetwalker, 2019-02-23:
H100i GT V2 is currently not supported?

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H100i GT V2
Firmware: 2.8.0.0
20 08
Temperature 0: 32.80 C
20 08
Temperature 1: 32.80 C
20 08
Temperature 2: 32.80 C
53
04 B0
Fan 0:  Mode 0x53
        Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
0B 22
Pump:   Mode 0x00
        Current/Max Speed 2850/0 RPM
DEBUG: deinit done

audiohacked, 2019-10-02:
Closing. Should be fixed.

closed 2019-10-02 by audiohacked

drumni, 2023-06-11:
include/driver.h:234:20: warning: ‘struct pump_control’ declared inside parameter list will not be visible outside of this definition or declaration
  234 |             struct pump_control* ctrl );
      |                    ^~~~~~~~~~~~
cc -std=c99 -Iinclude -g -I/usr/include/libusb-1.0  -DVERSION=\"v0.9.0.0-61d3\" -g -c -o lowlevel/asetek.o lowlevel/asetek.c
cc -std=c99 -Iinclude -g -I/usr/include/libusb-1.0  -DVERSION=\"v0.9.0.0-61d3\" -g -c -o lowlevel/commanderpro.o lowlevel/commanderpro.c
cc -std=c99 -Iinclude -g -I/usr/include/libusb-1.0  -DVERSION=\"v0.9.0.0-61d3\" -g -c -o lowlevel/coolit.o lowlevel/coolit.c
cc -std=c99 -Iinclude -g -I/usr/include/libusb-1.0  -DVERSION=\"v0.9.0.0-61d3\" -g -c -o lowlevel/platinum.o lowlevel/platinum.c
cc -std=c99 -Iinclude -g -I/usr/include/libusb-1.0  -DVERSION=\"v0.9.0.0-61d3\" -g -c -o lowlevel/rmi.o lowlevel/rmi.c
cc main.o device.o logic/options.o logic/options_fan.o logic/options_led.o logic/options_pump.o logic/print.o logic/scan.o logic/settings/commanderpro.o logic/settings/hydro_asetek.o logic/settings/hydro_asetekpro.o logic/settings/hydro_coolit.o logic/settings/hydro_platinum.o logic/settings/psu.o drivers/asetek.o drivers/asetekpro.o drivers/commanderpro.o drivers/coolit_old.o drivers/coolit.o drivers/dongle.o drivers/platinum.o drivers/rmi.o drivers/unsupported.o protocol/asetek/core.o protocol/asetek/fan.o protocol/asetek/led.o protocol/asetek/pump.o protocol/asetek/temperature.o protocol/asetekpro/core.o protocol/asetekpro/fan.o protocol/asetekpro/led.o protocol/asetekpro/pump.o protocol/asetekpro/temperature.o protocol/commanderpro/core.o protocol/commanderpro/fan.o protocol/commanderpro/power.o protocol/commanderpro/temperature.o protocol/coolit/core.o protocol/coolit/fan.o protocol/coolit/led.o protocol/coolit/pump.o protocol/coolit/temperature.o protocol/platinum/core.o protocol/platinum/crc.o protocol/platinum/fan.o protocol/platinum/fan_curve.o protocol/platinum/led.o protocol/platinum/pump.o protocol/platinum/temperature.o protocol/rmi/core.o protocol/rmi/power.o protocol/rmi/temperature.o protocol/rmi/time.o lowlevel/asetek.o lowlevel/commanderpro.o lowlevel/coolit.o lowlevel/platinum.o lowlevel/rmi.o -std=c99 -Iinclude -g -I/usr/include/libusb-1.0  -DVERSION=\"v0.9.0.0-61d3\" -lm -lusb-1.0  -o OpenCorsairLink.elf
[roggen@archlinux OpenCorsairLink]$ sudo ./OpenCorsairLink.elf
[sudo] password for roggen: 
Dev=0, CorsairLink Device Found: H100i GT V2!

[roggen@archlinux OpenCorsairLink]$ sudo ./OpenCorsairLink.el^C
[roggen@archlinux OpenCorsairLink]$ sudo ./OpenCorsairLink.elf --device 0              
Dev=0, CorsairLink Device Found: H100i GT V2!

Vendor: Corsair
Product: H100i GT V2
Firmware: 2.8.0.0
Temperature 0: 37.70 C
Temperature 1: 37.70 C
Temperature 2: 37.70 C
Fan 0:  Mode 0xA0
        Current/Max Speed 0/0 RPM
Pump:   Mode 0x00
        Current/Max Speed 1380/0 RPM

drumni, 2023-06-11:
> Closing. Should be fixed.

what update? i need to install for the fix