hyperkineticnerd/OpenCorsairLink #139: Needs GUI

notmutiny, 2018-09-29:
Should not take an hour and 5 webpages to turn my fan speeds to quiet. Application desperately needs a GUI given all of the problems other users have found when searching _how to quiet my fan speeds_.

6/10 thanks for the software but would not use again

audiohacked, 2018-09-29:
10/10 Not a real Issue.

closed 2018-09-29 by audiohacked

leavez, 2018-11-06:
Hi @audiohacked, would you like to provide a library interface, like libOpenCorsairLink, so others can build a GUI on it. 

I'm currently developing a GUI for Mac, and I have to call it through the command line and parse the output manually, which kinda inconvenient. I'm not familiar with the Makefile, maybe it's easy to implement.