hyperkineticnerd/OpenCorsairLink #142: H100i GT V2 fan current/max speed 0

corelema, 2018-11-09:
First of all, sorry if the question as already been answered, I searched but didn't find the solution

I am trying to change the fan mode as right now it's pretty noisy, but cannot figure why it doesn't work.

Here is the command I am trying:
`sudo ./OpenCorsairLink.elf --device 0 --pump mode=3 --fan channel=0,mode=3`

And here is the output:
```
Dev=0, CorsairLink Device Found: H100i GT V2!

Vendor: Corsair
Product: H100i GT V2
Firmware: 2.8.0.0
Temperature 0: 25.00 C
Temperature 1: 25.00 C
Temperature 2: 25.00 C
Fan 0:  Mode 0xDC
        Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
Pump:   Mode 0x03
        Current/Max Speed 2940/0 RPM
Setting fan to mode: 3
Setting pump to mode: 3
Unsupported Generic Pump Function
```
I am suspicious about the `Current/Max Speed 0/0 RPM`

Reading this thread https://github.com/audiohacked/OpenCorsairLink/issues/138 I was thinking that maybe my fan was indeed controlled by the motherboard, but I double checked, and the Q-fan control (Asus ROG motherboard) is disabled, so if I understand correctly, the AIO should receive full power.

Earlier I was messing up with lm-sensors to try to control all my fans, but I wiped it out (and installed it again, as it seems that it was a required dependency of mesa), so I guess it's back to the default. Maybe I do need to configure something somewhere to release its control (I'm on linux Arch).

Thanks!

ThoreauHenry, 2019-01-06:
Same issue on H80i GT V2.

netinetwalker, 2019-02-23:
H100i GT V2 is currently not supported?

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H100i GT V2
Firmware: 2.8.0.0
20 08
Temperature 0: 32.80 C
20 08
Temperature 1: 32.80 C
20 08
Temperature 2: 32.80 C
53
04 B0
Fan 0:  Mode 0x53
        Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
0B 22
Pump:   Mode 0x00
        Current/Max Speed 2850/0 RPM
DEBUG: deinit done

JeremyCaron, 2019-02-28:
Seeing the same with mine (while I can hear the fan ramped up from my custom curve due to video encoding):

```
Vendor: Corsair
Product: H100i GT V2
Firmware: 2.8.0.0
25 06
Temperature 0: 37.60 C
25 06
Temperature 1: 37.60 C
25 06
Temperature 2: 37.60 C
70
07 44
Fan 0:  Mode 0x70
        Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
0A 8C
Pump:   Mode 0x00
        Current/Max Speed 2700/0 RPM
DEBUG: deinit done
```

inkubux, 2019-09-14:
Same issue with the H80i GT V2.

Anybody has a fix.

Once I applied it I cannot revert to the settings that was set prior on windows.

jonasmalacofilho, 2019-09-17:
@audiohacked, I had noticed this as well, but because I have a non-Corsair/non-supported device, I didn't gave it much thought.

It seems the issue is related to [`corsairlink_asetek_fan_speed` saving the current fan speed to `fan_ctrl.speed`](https://github.com/audiohacked/OpenCorsairLink/blob/99e1d72fa5a0f6230f9b8805de663319f3e8f4fd/protocol/asetek/fan.c#L173), while [`hydro_asetek_settings` expects it in `fan_ctrl.speed_rpm`](https://github.com/audiohacked/OpenCorsairLink/blob/99e1d72fa5a0f6230f9b8805de663319f3e8f4fd/logic/settings/hydro_asetek.c#L85-L90).

Though I'm not sure which one should be used after all...  What should be the differences between these two fields?

audiohacked, 2019-09-22:
@jonasmalacofilho `fan_ctrl.speed` IMO should be used, but I think the `fan_ctrl.speed_rpm` still exists because some code still uses it. This just illustrates that I need to do a full code audit; when I get the time. Thanks for finding this.

closed 2019-10-02 by audiohacked

jonasmalacofilho, 2019-10-02:
@corelema 

Apart from the problem reporting the fan speed (which is now fixed), setting static PWM duty or rpm values is [not supported](https://github.com/audiohacked/OpenCorsairLink/blob/4234f4f258da3f50464aca182e41df78bb16b47b/driver.c#L218-L219) for this device.

But you can use stock (mode=[2-5]) or custom (mode=6) profiles:

```
sudo ./OpenCorsairLink.elf --device 0 --fan channel=0,mode=6,20:40,temps=20:40,speeds=25:100
```

PR #203 (2019-10-02): Fix where to read Hydro Asetek fan speeds from
Fixes: #142 ("H100i GT V2 fan current/max speed 0")

## Proposed changes

Struct fan_ctrl has both speed and speed_rpm fields.  The driver for Hydro Asetek devices stores the fan speed in the former, but we were accessing the latter.

This PR only fixes the Hydro Asetek driver.  While the speed_rpm field [might simply be some leftover from legacy code](), it is still used in other drivers and, from a quick glance, possibly correctly so.

## Types of changes

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking ch [... cut]
commit 4234f4f2 (2019-10-02): Fix where to read Hydro Asetek fan speeds from

Fixes: #142 ("H100i GT V2 fan current/max speed 0")