hyperkineticnerd/OpenCorsairLink #145: Add support for H100i Pro

JacksonCampolattaro, 2018-12-29:
I'd like to help add support for the H100i Pro. I noticed that the H115i Pro is supported, but the smaller version isn't recognized. Because the pump block appears to be identical between sizes, I'd imagine that the only appreciable difference between them is how they identify themselves over usb.

I'm not at my desktop right now, but I can post the output of lsusb tomorrow night.

I used this with the original H100i on my old computer, and I was really impressed with how straightforward it was compared to the official software. For now I'm stuck booting into windows to change settings on my new computer. 

I'm new to GitHub, but I'd love to contribute if there's any way I can help!  I'm planning to take a look at your process for adding support for the 280 mm version; if I get this to work by simply adding an extra device entry I'll be sure to submit a pull request.

JacksonCampolattaro, 2018-12-31:
I did some further testing, and it looks like the program already supports the H100i Pro! I'm resolving the issue, but it might be worth adding the 240mm version to the compatibility list since it seems to work.

Thank you for creating such a useful tool!

closed 2018-12-31 by JacksonCampolattaro

audiohacked, 2018-12-31:
@JacksonCampolattaro You could do a Pull request to add it to the README.md

JacksonCampolattaro, 2019-01-01:
That's a good idea!

I can confirm that pump and LED control works perfectly, but I'm running into trouble testing the fans. Both report Mode 0x00, and while the current rpm is accurate, max rpm is listed as zero. Trying to force the PWM, RPM or Channel returns 'Unsupported Generic Fan Function.' Setting a different fan mode appears to work fine, but the modes/speeds don't actually change. 

Is this a valid function call?
```
>  sudo ./OpenCorsairLink.elf --device 0 --fan mode=0,pwm=100 --debug

Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (046d:c52b)...
Checking USB device 2 (2752:0007)...
Checking USB device 3 (0d8c:0004)...
Checking USB device 4 (04d9:0198)...
Checking USB device 5 (1b1c:0c15)...
Corsair product detected. Checking if device is H100i Pro... Dev=0, CorsairLink Device Found: H100i Pro!
Checking USB device 6 (04d9:010b)...
Checking USB device 7 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H100i Pro
AA 12 34 01 00 04 00
Firmware: 1.0.4.0
hardware version returned: AB 12 34 02 00 00 00
A9 12 34 1C 03
Temperature 0: 28.03 C
00
41 12 34 00 01 C3
Fan 0:	Mode 0x00
	Current/Max Speed 451/0 RPM
00
41 12 34 01 01 9B
Fan 1:	Mode 0x00
	Current/Max Speed 411/0 RPM
function:corsairlink_asetekpro_pump_mode_read file: protocol/asetekpro/pump.c
pump mode response = 33 12 34 02
pump speed response = 31 12 34 0B 04
Pump:	Mode 0x02 (AsetekProPumpPerformance)
	Current/Max Speed 2820/0 RPM
Setting fan to mode: 0
Unsupported Generic Fan Function
DEBUG: deinit done
```
As soon as I've tested all the features I'll add the H100i Pro to the list of supported products.