hyperkineticnerd/OpenCorsairLink #146: Support for H100i RGB Ultimate

Teque5, 2018-12-31:
New H100i RGB Ultimate cooler, lsusb identified as 1b1c:0c18

I tried adding this device to device.c, copying 0x0c16:
```
{
.vendor_id = 0x1b1c,
.product_id = 0x0c18,
.device_id = 0xFF,
.name = "H100i RGB Ultimate",
.read_endpoint = 0x01 | LIBUSB_ENDPOINT_IN,
.write_endpoint = 0x01 | LIBUSB_ENDPOINT_OUT,
.driver = &corsairlink_driver_asetekpro,
.led_control_count = 1,
.fan_control_count = 2,
.pump_index = 0,
},
```
but it doesn't really seem to work:
```
$sudo ./OpenCorsairLink.elf --device=0 --debug
XXX
Checking USB device 8 (1b1c:0c18)...
Corsair product detected. Checking if device is H100i RGB Ultimate... Dev=0, CorsairLink Device Found: H100i RGB Ultimate!
XXXX

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H100i RGB Ultimate
Firmware: 0.0.0.0
00 00
Temperature 0:  0.00 C
00 00
Temperature 1:  0.00 C
00 00
Temperature 2:  0.00 C
00
00 00
Fan 0:	Mode 0x00
	Current/Max Speed 0/0 RPM
00
00 00
Fan 1:	Mode 0x00
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
00 00
Pump:	Mode 0x00
	Current/Max Speed 0/0 RPM
DEBUG: deinit done
```

Teque5, 2019-01-02:
No idea. I'm simply assuming the H100i would be like the H100i RGB. It may not be compatible. I tried all the drivers I found in `devices.c`.

audiohacked, 2019-01-06:
@Teque5 Can you upload or post a `lsusb -d 1b1c:0c18 -vvv`

Teque5, 2019-01-07:
Sure, but the device is clearly not recognized:
```
$ lsusb -d 1b1c:0c18 -vvv

Bus 001 Device 004: ID 1b1c:0c18 Corsair 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c18 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
```

audiohacked, 2019-01-07:
Looks like the H100i uses a Interrupt-type USB control interface, at least one-half of the protocol. Since lsusb didn't report what the other End Point was, I'm assuming its a control-type packet. I'm currently unable to do any reverse-engineering due to lack of money and hardware to work on this.

audiohacked, 2019-01-09:
@Teque5 Can you upload or post a `sudo lsusb -d 1b1c:0c18 -vvv` apparently lsusb didn't have full access to the USB as stated by the line: 'Couldn't open device, some information will be missing'.

Teque5, 2019-01-10:
my mistake:
```
$ sudo lsusb -d 1b1c:0c18 -vvv

Bus 001 Device 004: ID 1b1c:0c18 Corsair
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c18
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H100i Platinum
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
Device Status:     0x0001
  Self Powered
```

SarthakYadav, 2019-03-09:
Hello @audiohacked 
Is there work going on with supporting this cooler?
I'd like to help since I use this basically 24/7 and the 100% fan speed is atrocious.

H100i RGB Platinum has a **new** (supposedly) CoolIT pump, not an Asetek pump. It also has 24 addressable RGB LEDs (including 4 each in the fans). All other details have alluded me right now.

I've tried it with a lot of configurations with the CoolIT drivers. Do you think we'll need to rework the drivers for this cooler?

NullConvergence, 2019-05-08:
I'm also looking forward to this. @audiohacked can I help you do some reverse engineering?

rzrpt, 2019-06-10:
If it helps, here's my output:

`sudo lsusb -d 1b1c:0c18 -vvv

Bus 001 Device 009: ID 1b1c:0c18 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c18 
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H100i Platinum
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
Device Status:     0x0001
  Self Powered
`

TikiTDO, 2019-10-03:
I did a usbmon dump of the interaction between a H100i RGB Platinum, and the windows utility for managing it. 

This starts from connecting the driver to my VM, then I go through play a bit with different colors, and finally I try setting a bunch of fan modes, and finishes with me disconnecting the device from the VM.

[h100i_usb_dump.txt](https://github.com/audiohacked/OpenCorsairLink/files/3684714/h100i_usb_dump.txt)

I might come back at some point to poke at it, but this may be enough for someone with more experience to tell what's happening.

zatricky, 2019-11-25:
It seems the lsusb output from my H115i RGB Platinum is almost identical. Likely if we get support for the H100i it'll mean support for a couple others as well.

markubiak, 2019-11-25:
For anyone looking for a workaround while support for this is fleshed out, it's perfectly valid to just connect the fan headers to the motherboard instead of the water block. I did this to quiet down the fans and have had no problems with temperature.

Legion2, 2020-01-01:
I think the H100i PLATINUM, H115i PLATINUM and H100i PLATINUM SE all use the same protocol as the Commander PRO.

audiohacked, 2020-01-01:
@Legion2 I can personally vouch that the H100i Platinum and the Commander Pro don't share the same protocol.

Legion2, 2020-01-01:
But they are very similar. I can flash an Arduino with the Commander Pro firmware and use the H100i Platinum USB ids and iCUE thinks the Arduino is a H100i Platinum and I can see the commands send by iCUE on the arduino.

audiohacked, 2020-01-01:
What do you mean "very similar"? Can your Arduino correctly interpret and feedback **all** the statuses and sensors as expected of iCUE?

Legion2, 2020-01-01:
No, it can only handle the Commander Pro commands correctly, but the structure of the H100i Platinum protocol is similar, only the commands them self differ.

gitcoinbot, 2020-03-20:
Issue Status: 1. **Open** 2. Started 3. Submitted 4. Done 

<hr>

__This issue now has a funding of 0.43 ETH (50.35 USD @ $117.09/ETH)  attached to it.__

 * If you would like to work on this issue you can 'start work' [on the Gitcoin Issue Details page](https://gitcoin.co/issue/audiohacked/OpenCorsairLink/146/4141).
* Want to chip in? Add your own contribution [here](https://gitcoin.co/issue/audiohacked/OpenCorsairLink/146/4141).
* Questions? Checkout <a href='https://gitcoin.co/help'>Gitcoin Help</a> or the <a href='https://gitcoin.co/slack'>Gitcoin Slack</a>
* $79,042.29 more funded OSS Work available on the [Gitcoin Issue Explorer](https://gitcoin.co/explorer)

gitcoinbot, 2020-03-22:
Issue Status: 1. Open 2. **Started** 3. Submitted 4. Done 

<hr>

__Work has been started__.

These users each claimed they can complete the work by 6 months from now.
Please review their action plans below:

**1) [shnqdr](https://gitcoin.co/shnqdr) has started work.**

ShirinGhaderi  need the wallet

Learn more [on the Gitcoin Issue Details page](https://gitcoin.co/issue/audiohacked/OpenCorsairLink/146/4141).

tkruger206, 2020-03-23:
Super excited to get this for my Hackintosh!

gitcoinbot, 2020-04-03:
Issue Status: 1. Open 2. **Started** 3. Submitted 4. Done 

<hr>

__Work has been started__.

These users each claimed they can complete the work by 5 months, 2 weeks from now.
Please review their action plans below:

**1) [poorna123613](https://gitcoin.co/poorna123613) has started work.**

I can solve query by understanding the query and providing the right intructions.

Learn more [on the Gitcoin Issue Details page](https://gitcoin.co/issue/audiohacked/OpenCorsairLink/146/4141).

makk50, 2020-05-29:
Hi, I have tested your latest changes, and now i can read the cooling liquid temperature on my h115i platinum:
marco@pc-kubuntu:/datos/OpenCorsairLink-testing$ sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H115i Platinum!

Vendor: Corsair
Product: H115i Platinum
Firmware: 1.1.15
Temperature 0: 30.18 C
Fan 0:  Unknown (STUB)
        Current/Max Speed 296/65535 RPM
Fan 1:  Unknown (STUB)
        Current/Max Speed 296/65535 RPM
Pump:   Mode 0x00

It seems you are getting closer to provide support for CoolIT AIOs. Great work!
If i can help you with my limited knowledge, I can test, or provide you info if you need

cometta, 2020-07-22:
guys, anyone knows a way to off the rgb colors on linux?

gitcoinbot, 2020-09-19:
Issue Status: 1. Open 2. **Cancelled** 

<hr>

__The funding of 0.43 ETH  (163.02 USD @ $379.11/ETH) attached to this issue has been **cancelled** by the bounty submitter__

 
* Questions? Checkout <a href='https://gitcoin.co/help'>Gitcoin Help</a> or the <a href='https://chat.gitcoin.co/'>Gitcoin Chat</a>
* $760,745.98 more funded OSS Work available on the [Gitcoin Issue Explorer](https://gitcoin.co/explorer)

ShnQdr, 2020-09-22:
Cool what shall I do?

TikiTDO, 2020-09-22:
The [liquidctl](https://github.com/jonasmalacofilho/liquidctl) project now supports this device, so this issue can be closed.

closed 2020-09-22 by audiohacked