hyperkineticnerd/OpenCorsairLink #148: Support for the H115i RGB Platinum

TJBrown392, 2019-01-14:
I just recently purchased the H115i RGB Platinum for my system and was looking for a way to control it through Linux. I ran across this project and thought I'd give it a try, but it doesn't appear to be supported yet. I'd be willing to provide as much information as I possible can. 

Here is the lsusb output of the device: 
`Bus 001 Device 005: ID 1b1c:0c17 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c17 
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H115i Platinum
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
Device Status:     0x0001
  Self Powered
`

If this can be merged with the issue related to the H100i RGB Platinum let me know so I can post this information over there instead.

airbat, 2019-04-21:
+1 we need it please. In i can help in anyway ... tell me.

PeterJahnes, 2019-07-02:
+1 as well. I have made changes to device.c on my local machine and got it to recognize the device, but it can not read or write values. If someone could point me in the right direction on how to find the proper settings, I will do the work and make a pull request.

Jiivee1, 2019-07-04:
+1 as well.

Looks to me Corsair has changed H115i RGB Platinum to look like a keyboard instead of their previous interface.

audiohacked, 2019-07-04:
After a quick Google search, I read that the H115i Platinum is manufactured by CoolIT. So **maybe** it might work with the coolit protocol.

PeterJahnes, 2019-07-04:
When I was comparing the lsusb output from the H115i platinum with others like H100i, they look very different. Thanks for the CooIIT tip, Ill give that a go later today or tomorrow and see what I can get.

Any idea on led count? looks like there are a lot of them on this device.

Jiivee1, 2019-07-06:
I tested with CoolIT driver. It can dump some data and it seems to be different every time :(

Here is lsusb output:
`# lsusb -d 1b1c:0c17 -vvv

Bus 001 Device 003: ID 1b1c:0c17 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c17 
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H115i Platinum
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered`

As H115i RGB Platinum pretends to be a keyboard, does it actually need its own driver? Snipped I have now in device.c is below. I have first tried to get control of pump and later of fans and leds.

`{
        .vendor_id = 0x1b1c,
[... 106 lines cut ...]
DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H115i RGB Platinum
Firmware: 0.0.0
Fan count: 19
Fan 0:  Not connected or failed
        Current/Max Speed 0/0 RPM
Fan 1:  Not connected or failed
        Current/Max Speed 0/0 RPM
Fan 2:  Not connected or failed
        Current/Max Speed 0/0 RPM
Fan 3:  Not connected or failed
        Current/Max Speed 1535/14130 RPM
Fan 4:  Not connected or failed
        Current/Max Speed 0/0 RPM
Fan 5:  Not connected or failed
        Current/Max Speed 0/0 RPM
Fan 6:  Not connected or failed
        Current/Max Speed 0/0 RPM
Pump:   Mode 0x00
        Current/Max Speed 1535/14130 RPM
DEBUG: deinit done`

Jiivee1, 2019-07-06:
With help of [usbmon](https://www.kernel.org/doc/Documentation/usb/usbmon.txt) documentation I got some debugging output.

Command and its output:
```
# OpenCorsairLink.elf --dump --device=0
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (099a:6200)...
Checking USB device 2 (1d6b:0002)...
Checking USB device 3 (1d6b:0003)...
Checking USB device 4 (1d6b:0002)...
Checking USB device 5 (1d6b:0003)...
Checking USB device 6 (1d6b:0002)...
Checking USB device 7 (1d6b:0003)...
Checking USB device 8 (8087:0aa7)...
Checking USB device 9 (1b1c:0c17)...
Corsair product detected. Checking if device is H115i RGB Platinum... Dev=0, CorsairLink Device Found: H115i RGB Platinum!
Checking USB device 10 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H115i RGB Platinum
Firmware: 0.0.0
Temperature sensors count: 0
Fan count: 255
Limiting reading to 7 first fans.
Fan 0:  Not connected or failed
        Current/Max Speed 0/0 RPM
Fan 1:  Not connected or failed
        Current/Max Speed 1535/14130 RPM
Fan 2:  Not connected or failed
        Current/Max Speed 0/0 RPM
Fan 3:  Not connected or failed
        Current/Max Speed 0/0 RPM
Fan 4:  Not connected or failed
        Current/Max Speed 0/0 RPM
Fan 5:  Not connected or failed
        Current/Max Speed 1535/14130 RPM
Fan 6:  Not connected or failed
        Current/Max Speed 0/0 RPM
Pump:   Mode 0x00
        Current/Max Speed 0/0 RPM
DEBUG: deinit done
```

Usbmon output:
```
ffff8f7a6b2fb0c0 1968920381 S Co:1:003:0 s 21 09 0200 0000 0040 64 = 03810700 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff8f7a6b2fb0c0 1968921856 C Co:1:003:0 0 64 >
ffff8f7a6b2fb0c0 1968921908 S Ii:1:003:1 -115:4 64 <
ffff8f7a6b2fb0c0 1969922182 C Ii:1:003:1 -2:4 0
ffff8f7a6b2fb0c0 1969922278 S Co:1:003:0 s 21 09 0200 0000 0040 64 = 03820901 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff8f7a6b2fb0c0 1969923969 C Co:1:003:0 0 64 >
ffff8f7a6b2fb0c0 1969924260 S Ii:1:003:1 -115:4 64 <
ffff8f7a6b2fb0c0 1970924487 C Ii:1:003:1 -2:4 0
ffff8f7a6b2fb0c0 1970924537 S Co:1:003:0 s 21 09 0200 0000 0040 64 = 0381070d 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff8f7a6b2fb0c0 1970926074 C Co:1:003:0 0 64 >
ffff8f7a6b2fb0c0 1970926120 S Ii:1:003:1 -115:4 64 <
ffff8f7a6b2fb0c0 1971926346 C Ii:1:003:1 -2:4 0
ffff8f7a6b2fb0c0 1971926388 S Co:1:003:0 s 21 09 0200 0000 0040 64 = 03810711 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff8f7a6b2fb0c0 1971928168 C Co:1:003:0 0 64 >
ffff8f7a6b2fb0c0 1971928342 S Ii:1:003:1 -115:4 64 <
ffff8f7a6b2fb0c0 1972928548 C Ii:1:003:1 -2:4 0
ffff8f7a6b2fb0c0 1972928597 S Co:1:003:0 s 21 09 0200 0000 0040 64 = 07820610 00830712 00000000 00000000 00000000 00000000 00000000 00000000
ffff8f7a6b2fb0c0 1972930284 C Co:1:003:0 0 64 >
ffff8f7a6b2fb0c0 1972930457 S Ii:1:003:1 -115:4 64 <
ffff8f7a6b2fb0c0 1973930685 C Ii:1:003:1 -2:4 0
ffff8f7a6b2fb0c0 1973930723 S Co:1:003:0 s 21 09 0200 0000 0040 64 = 0a840610 00850916 86091700 00000000 00000000 00000000 00000000 00000000
[... 41 lines cut ...]
ffff8f7a6b2fb0c0 1982958362 S Ii:1:003:1 -115:4 64 <
ffff8f7a6b2fb0c0 1982962403 C Ii:1:003:1 0:4 64 = ffa81400 ff05002d 32374100 00000000 0000e803 2d1d5500 e8032d1d 55010000
ffff8f7a6b2fb0c0 1982962445 S Co:1:003:0 s 21 09 0200 0000 0040 64 = 07a00610 06a10712 00000000 00000000 00000000 00000000 00000000 00000000
ffff8f7a6b2fb0c0 1982964455 C Co:1:003:0 0 64 >
ffff8f7a6b2fb0c0 1982964489 S Ii:1:003:1 -115:4 64 <
ffff8f7a6b2fb0c0 1983964714 C Ii:1:003:1 -2:4 0
ffff8f7a6b2fb0c0 1983964745 S Co:1:003:0 s 21 09 0200 0000 0040 64 = 0aa20610 06a30916 a4091700 00000000 00000000 00000000 00000000 00000000
ffff8f7a6b2fb0c0 1983966442 C Co:1:003:0 0 64 >
ffff8f7a6b2fb0c0 1983966469 S Ii:1:003:1 -115:4 64 <
ffff8f7a6b2fb0c0 1984966696 C Ii:1:003:1 -2:4 0
ffff8f7a6b2fb0c0 1984966742 S Co:1:003:0 s 21 09 0200 0000 0040 64 = 07810610 04820712 00000000 00000000 00000000 00000000 00000000 00000000
ffff8f7a6b2fb0c0 1984968523 C Co:1:003:0 0 64 >
ffff8f7a6b2fb0c0 1984968549 S Ii:1:003:1 -115:4 64 <
ffff8f7a6b2fb0c0 1985968768 C Ii:1:003:1 -2:4 0
ffff8f7a6b2fb0c0 1985968797 S Co:1:003:0 s 21 09 0200 0000 0040 64 = 0a830610 04840916 85091700 00000000 00000000 00000000 00000000 00000000
ffff8f7a6b2fb0c0 1985970653 C Co:1:003:0 0 64 >
ffff8f7a6b2fb0c0 1985970680 S Ii:1:003:1 -115:4 64 <
ffff8f7a6b2fb0c0 1986970906 C Ii:1:003:1 -2:4 0
ffff8f7a6b2fb0c0 1986971029 S Co:1:003:0 s 21 0a 0000 0000 0000 0
ffff8f7a6b2fb0c0 1986971965 C Co:1:003:0 -32 0
ffff8f7a6b2fb0c0 1986971974 S Ci:1:003:0 s 81 06 2200 0000 001b 27 <
ffff8f7a6b2fb0c0 1986974748 C Ci:1:003:0 0 27 = 0600ff09 01a10115 0026ff00 75089540 09018102 95400901 9102c0
```

I can test this further as needed.

ajmi982, 2019-07-27:
You all seem to really know your stuff. I have a h115i platinum that I want to flash firmware to without using iCUE because iCUE doesn’t recognize the AIO yet windows and USDeview does. I believe it got slightly bricked on a FW update and now iCUE refuses to find it.

DNavaratne, 2019-09-23:
@Jiivee1 have you had any luck getting H115i RGB Platinum working with OpenCorsairLink ?

audiohacked, 2019-10-03:
Closing this until I can get hardware to support this.

closed 2019-10-03 by audiohacked

thekev, 2020-05-13:
There's some reverse-engineering work going on over here https://github.com/jonasmalacofilho/liquidctl/issues/76