hyperkineticnerd/OpenCorsairLink #150: H110i GTX not showing temps / speeds

broyles, 2019-01-21:
I am having issues showing my H110i, GTX I am sure it used to work, but on the latest version all I am getting is 0's from it!

I am using Fedora 29 with kernel 4.19.15-300.fc29.x86_64

Here is the outputs, any suggestions ?

sudo ./OpenCorsairLink.elf --device 1 --debug
Checking USB device 0 (8087:8002)...
Checking USB device 1 (1d6b:0002)...
Checking USB device 2 (1d6b:0003)...
Checking USB device 3 (1b1c:1c0d)...
Corsair product detected. Checking if device is RM1000i... Dev=0, CorsairLink Device Found: RM1000i!
Checking USB device 4 (1b1c:0c07)...
Corsair product detected. Checking if device is H110i GTX... Dev=1, CorsairLink Device Found: H110i GTX!
Checking USB device 5 (1d6b:0002)...
Checking USB device 6 (8087:800a)...
Checking USB device 7 (1d6b:0002)...
Checking USB device 8 (174c:3074)...
Checking USB device 9 (174c:3074)...
Checking USB device 10 (1d6b:0003)...
Checking USB device 11 (174c:2074)...
Checking USB device 12 (0b05:180a)...
Checking USB device 13 (1b1c:1b11)...
Checking USB device 14 (046d:c07d)...
Checking USB device 15 (174c:2074)...
Checking USB device 16 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 1
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H110i GTX
Firmware: 0.0.0.0
00 00
Temperature 0:  0.00 C
00 00
Temperature 1:  0.00 C
00 00
Temperature 2:  0.00 C
00
00 00
Fan 0:  Mode 0x00
        Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
00 00
Pump:   Mode 0x00
        Current/Max Speed 0/0 RPM
DEBUG: deinit done

sudo lsusb -d 1b1c:0c07 -v

Bus 005 Device 002: ID 1b1c:0c07 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c07 
  bcdDevice            1.00
  iManufacturer           1 Corsair Components, Inc.
  iProduct                2 H110iGTX Cooler
  iSerial                 3 7289_1.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

augustopaulogh, 2019-02-25:
Hi, 
I'm on macOS 10.14.3 (18D109) and here his my output for my 2 Corsair devices:
```
sudo ./OpenCorsairLink.elf --debug
Checking USB device 0 (05ac:8290)...
Checking USB device 1 (046d:c077)...
Checking USB device 2 (046d:c315)...
Checking USB device 3 (1b1c:0c07)...
Corsair product detected. Checking if device is H110i GTX... Dev=0, CorsairLink Device Found: H110i GTX!
Checking USB device 4 (1b1c:1c07)...
Corsair product detected. Checking if device is HX1000i... Dev=1, CorsairLink Device Found: HX1000i!
Checking USB device 5 (046d:082d)...
Checking USB device 6 (05e3:0606)...
Checking USB device 7 (0b0e:0312)...
Checking USB device 8 (05e3:0610)...
Checking USB device 9 (05e3:0608)...
Checking USB device 10 (05e3:0608)...
Checking USB device 11 (0bda:5411)...
Checking USB device 12 (048d:8297)...
Checking USB device 13 (05e3:0608)...

DEBUG: scan done, start routines
DEBUG: device_number = -1
```
But neither the H110i GTX nor the HX1000i return any device status data:
```
sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H110i GTX!
Dev=1, CorsairLink Device Found: HX1000i!

Vendor: Corsair
Product: H110i GTX
Firmware: 0.0.0.0
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Fan 0:	Mode 0x00
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
Pump:	Mode 0x00
	Current/Max Speed 0/0 RPM
```

```
sudo ./OpenCorsairLink.elf --device 1
Dev=0, CorsairLink Device Found: H110i GTX!
Dev=1, CorsairLink Device Found: HX1000i!

Vendor: 
Product: 
Firmware: NA
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Powered: 0 (0d.  0h)
Uptime: 0 (0d.  0h)
Supply Voltage:  0.00 V
Total Watts:  0.00 W
Output 12v:
	Voltage  0.00 V
	Amps  0.00 A
	Watts  0.00 W
Output 5v:
	Voltage  0.00 V
	Amps  0.00 A
	Watts  0.00 W
Output 3.3v:
	Voltage  0.00 V
	Amps  0.00 A
	Watts  0.00 W
```

karmitage16, 2019-04-11:
I am also experiencing this on the following Manjaro Linux system: 4.19.32-1-MANJARO
Cinnamon Version 4.0.10

sudo opencorsairlink --device 0
Dev=0, CorsairLink Device Found: H110i GTX!

Vendor: Corsair
Product: H110i GTX
Firmware: 0.0.0.0
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Fan 0:	Mode 0x00
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
Pump:	Mode 0x00
	Current/Max Speed 0/0 RPM

Rehaell, 2019-05-12:
+1 with the same issue.

Using Ubuntu 18.04 LTS.

```Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (1d6b:0002)...
Checking USB device 2 (1d6b:0003)...
Checking USB device 3 (1532:0203)...
Checking USB device 4 (1038:170e)...
Checking USB device 5 (1a40:0101)...
Checking USB device 6 (0a12:0001)...
Checking USB device 7 (1038:1294)...
Checking USB device 8 (1038:1290)...
Checking USB device 9 (0451:2036)...
Checking USB device 10 (1b1c:0c07)...
Corsair product detected. Checking if device is H110i GTX... Dev=0, CorsairLink Device Found: H110i GTX!
Checking USB device 11 (28de:1142)...
Checking USB device 12 (1d6b:0002)...
Checking USB device 13 (1d6b:0003)...
Checking USB device 14 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H110i GTX
Firmware: 0.0.0.0
00 00
Temperature 0:  0.00 C
00 00
Temperature 1:  0.00 C
00 00
Temperature 2:  0.00 C
00
00 00
Fan 0:	Mode 0x00
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
00 00
Pump:	Mode 0x00
	Current/Max Speed 0/0 RPM
DEBUG: deinit done
```

jhd, 2019-05-26:
+1 for this issue. My output is the same as @Rehaell except I'm using Ubuntu 19.04.

jhd, 2019-06-02:
The first libusb command at lowlevel:asetek.c:38 returns LIBUSB_ERROR_PIPE
    rr = libusb_control_transfer( dev_handle, 0x40, 0x00, 0xffff, 0x0000, NULL, 0, 0 );

The subsequent libusb call returns no error and masks the return code from the first call.

audiohacked, 2019-06-03:
The problem seems to be a matter of access rights or LibUSB problem.

audiohacked, 2019-06-19:
Looks like I didn't catch that this issue is a duplicate of #134

closed 2019-06-19 by audiohacked