hyperkineticnerd/OpenCorsairLink #151: HX1200i make install and led colour don't work

FrancescoBorzi, 2019-01-30:
**Describe the bug**

After `make`, I did `sudo make install` and I got:

> make: *** No rule to make target 'install'. Stop.

I tried to use this command to change colour:

```
sudo ./OpenCorsairLink.elf --device 0 --led channel=0,mode=0,color=0000CD
```

but the colour remains the same

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: PC
 - OS: Ubuntu 18.04
 - Browser: Chrome (but why does it matter?)
 - Version: `OpenCorsairLink Version: v0.9.0.0Dev=0, CorsairLink Device Found: HX1200i!`

closed 2019-01-31 by audiohacked

FrancescoBorzi, 2019-02-02:
@audiohacked you could have mentioned why the issue report is not valid and how I can make it right

audiohacked, 2019-02-02:
@FrancescoBorzi The HX1200i doesn't have LEDs.

audiohacked, 2019-02-02:
There is no install....

FrancescoBorzi, 2019-02-02:
@audiohacked then we should update the `README.md`, currently saying:

[image]

audiohacked, 2019-02-02:
The README.md is correct, just no install support in Make.

bmccord2, 2019-02-26:
@audiohacked This answer confuses me.  If there is no target defined in the make file named "install" how is the Makefile correct?