hyperkineticnerd/OpenCorsairLink #152: My H115i AIO is now being misreported as H110i GT V2

nick-s-b, 2019-02-04:
**Describe the bug**
Hi,
We've been discussing how to add AIO fan/pump/temp readings to lm-sensors over at [lm-sensors issue #118](https://github.com/lm-sensors/lm-sensors/issues/118) and @jameshilliard noticed that OpenCorsairLink misreported the model of his AIO. After checking it myself, the same issue was affecting my install as well.

Not sure how recent this bug is but OCL did report my AIO correctly in Aug 2018 as evidenced in this post: https://github.com/audiohacked/OpenCorsairLink/issues/120#issuecomment-410540086

````
$ sudo OpenCorsairLink --version
OpenCorsairLink Version: v0.9.0.0Dev=0, CorsairLink Device Found: H110i GT V2!
````

````
$ lsusb -d 1b1c: -v

Bus 002 Device 005: ID 1b1c:0c0a Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c0a 
  bcdDevice            1.00
  iManufacturer           1 Corsair Components, Inc.
  iProduct                2 H115i
  iSerial                 3 7289_2.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
````

jameshilliard, 2019-02-04:
The behavior of the official corsair software does appear to use the product id `0x0c0a` to identify the protocol, it seems however that the displayed name is separate.

Miziak, 2019-04-03:
@nick-s-b I have a question base control of fans working well? This issue is only about reported name? I want switch to linux and have this cooler and I'm worried about it's support can You tell me it's fine to use it?
Sorry for offtopic

nick-s-b, 2019-04-03:
@Miziak The issue is only the name. Everything else should work fine. Just switch :)

audiohacked, 2019-06-19:
I think originally the H115i was/is actually a H110i GT V2. I assume its a branding change.

tad1073, 2019-06-24:
Same issue.

audiohacked, 2019-06-25:
If anyone wants this fixed, please submit a patch. It's really easy.

closed 2019-10-03 by audiohacked