hyperkineticnerd/OpenCorsairLink #153: Fan RPM control for Commander Pro

mtalpine, 2019-02-07:
Thanks for all of your hard work on this. It is greatly appreciated!

I was hoping to be able to at least set the RPM of the fans that are attached to the Commander PRO device.

This is the current readout of Dev=1, CorsairLink Device Found: Commander PRO!
Vendor: Corsair
Product: Commander PRO
Firmware: V0.8.210
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Temperature 3:  0.00 C
Output 12v: 12.08 V
Output 5v:  5.00 V
Output 3.3v:  3.35 V

I found a person who developed something for FreeNAS at this link: https://forums.freenas.org/index.php?threads/opencorsairlink-in-a-jail-to-control-fans.71873/ They were able to implement a perl file and some additional changes to get a configuration that looks like this:

root@FreeNAS:/mnt/vol4/scripts # ./OpenCorsairLink.elf --device 0 --fan channel=0,mode=0
Dev=0, CorsairLink Device Found: Commander PRO!
Vendor: Corsair
Product: Commander PRO
Firmware: V0.5.180
Temperature 0: 33.00 C
Temperature 1: 35.00 C
Temperature 2: 36.00 C
Temperature 3: 34.00 C
Output 12v: 12.13 V
Output 5v:  5.01 V
Output 3.3v:  3.38 V
FAN PWM 1: 55 %
FAN 1: 1243
FAN PWM 2: 55 %
FAN 2: 1203
FAN PWM 3: 45 %
FAN 3: 1422
FAN PWM 4: 45 %
FAN 4: 1426
FAN PWM 5: 45 %
FAN 5: 1424
FAN PWM 6: 100 %
FAN 6: 0

I have tried to dig into this a little bit but I am new to programming and not able to figure out enough of it to see if I can do something similar to OCL for my Ubuntu system. Any chance of having at least the ability to set the speeds of the fans that are connected to the Commander PRO? Thanks again!

mtalpine, 2019-02-09:
Something I just learned is that the Commander Pro will respond to the included temperature sensors even when in BIOS and Linux. I installed one and updated my custom profile in ICUE while in Windows to use the sensor. Then when I rebooted into Ubuntu the fans no longer ran at 100%. This is a great fix for those trying to slow down the fans in Linux. Sorry for yet another post but I hope this helps anyone else who is looking for a fix to the fan speed issue. Thanks again for all of your hard work on OCL!

nightah, 2019-03-05:
@audiohacked could the Commander Pro have been broken with a firmware update?

I just recently purchased one and notice the same issue as above, I cannot control any of the fans despite the fact that I know the unit is working and can be adjusted in Windows with either CorsairLink or iCUE.

Any instance of a person seemingly running OpenCorsairLink and actually being able to read or adjust fan speeds seems to suggest older firmware being used.

closed 2019-04-09 by audiohacked

Grasan, 2019-04-11:
I can adjust the fan speed, but only stopping the fans and i cant change back to the previous setting.
If I write <./OpenCorsairLink.elf --device 0 --fan channel=n, mode=3> the selected fan stops spinning completely. And good luck changing it back because that's impossible in the terminal. I have to restart the computer for the fans to reset back to the previous speeds.

nightah, 2019-04-12:
Only `--mode=0` and `--mode=1` have been implemented so far, i.e: Fixed PWM and Fixed RPM.

PR #161 (2019-04-09): Commander Pro Fan control
## Proposed changes

This change allows users with the Corsair Commander Pro to read and configure any attached fan's via either specified RPM or PWM %.

This fixes #153. 

## Types of changes

What types of changes does your code introduce to Appium?
_Put an `x` in the boxes that apply_

- [ ] Bugfix (non-breaking change which fixes an issue)
- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

## Checklist

_Put an `x` in the boxes that apply. You can  [... cut]
commit 985f6efc (2019-04-09): Merge pull request #161 from antilax-3/feature/commander_pro_fan

Commander Pro Fan control
Fixes #153