hyperkineticnerd/OpenCorsairLink #155: Merge with ckb-next project ?

mfp20, 2019-02-18:
Hi,

I've multiple corsair devices and I'd like to have a similar - and better - experience to the one it is possible to obtain on windows. In order to get my result - LEDs wise - there's the need to manage all corsair devices using a single profile manager. In a similar way to iCUE for Windows.
By looking at linux kernel I guessed there was the need to develop some new modules; so in the past 2 days I started to work on kernel modules. But then I figured out that ckb-next was using usbfs instead, moving the corsair protocol to userland thanks to ckb-daemon.
So I guessed that moving to userland might be safer than a kernel module, I dropped my kernel modules development, and start digging into the current projects:
1) OpenCorsairLink: PSUs, Commander Pro, Lighting Node Pro, ... .
2) ckb-next: keyboards, mices, ST100 (audio device and headphones stand).

Merging those two into a single 'libcorsair' (and daemon, cli, gui) would allow to use a single manager to manage (and sync) also similar products from other brands (ex: logitech, asus, roccat, razer, ...).
I'd be happy to work on it but I can't/don't want to spawn a new project.

Is there a chance for the developers of both projects to work together on a single libcorsair (and daemon, cli, gui)?

A few extra notes:
1) MacOSX support can be dropped ( http://forum.corsair.com/v3/showpost.php?p=982647&postcount=1 ).
2) Corsair won't do a linux manager in a near future ( http://forum.corsair.com/v3/showpost.php?p=982788&postcount=4 ).
3) Some extra sysfs love might be useful: temp&fans interfaces for lm_sensors (/sys/class/hwmon ?), have a look at https://github.com/lm-sensors/lm-sensors/issues/163#issuecomment-460100556 ; LEDs interface (/sys/class/leds ?).

cheers

audiohacked, 2019-02-18:
I'd like to note that iCUE doesn't implement full functionality for CLINK devices. And the support on Mac OS X is even smaller than the Windows version.

mfp20, 2019-02-18:
> I'd like to note that iCUE doesn't implement full functionality for CLINK devices.

Good to know. It would be much much easier to hook jtag/ttl cables and rework the firmware itself. But I don't want to void the warranty opening the devices; so, I don't really know what's possible and what's not.
I just bought corsair stuff 'cause I've spotted your projects (for linux), and the protocol reverse engineering on corsair's forums. Even on windows current LINK and iCUE software are pretty underachieving. There's really need of a good software suite for this magnificent HW.

> And the support on Mac OS X is even smaller than the Windows version.

A few ifdefs in the 'libcorsair' code aren't a problem, given the libusb compatibility layer.

As I was saying in the other thread (@ckb-next [twin issue](https://github.com/ckb-next/ckb-next/issues/376)) both Windows and OSX have companies behind, and Corsair is a company as well. So their ecosystem is not my business. My business is my system only. Plus, I'm happy to pack together with other developers in order to support the rest of the linux users.
No Free development for companies. I'm even a bit upset because some of the linux developers make business out of it: once the development is distributed, all or none should get money out of it. Other things are frustrating. Anyway, politics a part... 

I'm currently investigating the idea of making a dumb linux module that read, parse, and eventually duplicate the traffic in order to feed multiple char devices. A kind of router/proxy.
In this way it is possible to feed both hwmon+leds and usbfs; this allow for cross-platform applications (ie: based on libusb) and not (ie: lm_sensors).

closed 2019-04-09 by audiohacked