hyperkineticnerd/OpenCorsairLink #156: Custom Fan Control - now implemented?

seabbs, 2019-02-27:
Hi, 

Thanks for implementing/supporting this. Shocking that Corsair haven't released a version for Linux. 

I was just wondering if custom curves are working correctly. On my system (using a H100i v2) I get a continuous load when  I set a custom curve? Just wanted to clarify if this is currently an implemented feature and this is a bug or if it hasn't yet been implemented.

On a related note are the preset modes (i.e balanced, performance etc) just a single fan setting regardless of temp? 

Thanks, 

Sam

JeremyCaron, 2019-02-28:
I believe I was just able to set a custom curve doing something like this:

sudo OpenCorsairLink --device 0 --fan channel=1,mode=6,temps=0:35:36:37:38:40,speeds=0:45:55:60:75:100

However, this seems to scale with the liquid temps rather than CPU temps, which I guess makes sense the more I think about it.  However, I could swear I could scale with CPU temps in Windows.  I'll check on that at some point and report back.

seabbs, 2019-03-04:
Interesting. I was using something comparable and getting full fan speed all the time.

audiohacked, 2019-10-02:
The fan scaling using CPU is because CorsairLink in Windows runs as a service. Closing this due to differences between Operating Systems.

closed 2019-10-02 by audiohacked