hyperkineticnerd/OpenCorsairLink #157: Corsair Hydro H100i Pro RGB

robot110011, 2019-03-15:
**Describe the bug**
Changing fan mode not work. I tried to change to quiet mode and the fan are spinning exactly the same way as before

**To Reproduce**
Steps to reproduce the behavior:
sudo ./OpenCorsairLink.elf --device 0 --fan channel=0,mode=3 

**Expected behavior**
It should reduce to ~1000rpm

**Screenshots**

debug:
```
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (0b05:18a3)...
Checking USB device 2 (0b05:17cb)...
Checking USB device 3 (05e3:0610)...
Checking USB device 4 (258a:1007)...
Checking USB device 5 (1b1c:0c15)...
Corsair product detected. Checking if device is H100i Pro... Dev=0, CorsairLink Device Found: H100i Pro!
Checking USB device 6 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H100i Pro
AA 12 34 01 00 04 00
Firmware: 1.0.4.0
hardware version returned: AB 12 34 02 00 00 00
A9 12 34 24 06
Temperature 0: 36.06 C
00
41 12 34 00 05 BE
Fan 0:  Mode 0x00
        Current/Max Speed 1470/0 RPM
00
41 12 34 01 05 A0
Fan 1:  Mode 0x00
        Current/Max Speed 1440/0 RPM
function:corsairlink_asetekpro_pump_mode_read file: protocol/asetekpro/pump.c
pump mode response = 33 12 34 02
pump speed response = 31 12 34 0B 04
Pump:   Mode 0x02 (AsetekProPumpPerformance)
        Current/Max Speed 2820/0 RPM
Setting fan to mode: 3
DEBUG: deinit done
robot@robot-machine:~/
```

**Desktop (please complete the following information):**
 - OS: Neon KDE (kernel 4.15)
 - Browser chrome
 - Version 73

**Additional context**
I also tried with another channels (1-4) and got same behaviour

metthal, 2019-03-18:
I have the same cooling and what I found out when playing around with sniffing iCUE communication, this is what iCUE sends when I set the custom curve profile for my fans:

```
0000   1b 00 f0 9a 5e 22 0d e5 ff ff 00 00 00 00 09 00
0010   00 01 00 02 00 01 03 10 00 00 00 40 00 1e 20 22
0020   25 28 2b 30 14 19 22 32 41 51 5f
```

The important part here is 

```
40 00 1e 20 22 25 28 2b 30 14 19 22 32 41 51 5f
```

If I understand it correctly

`0x40` = command for fan RPM change most probably
`0x00` = seems like fan channel/index because this same data were sent two times, first time with `0x00` and second time with `0x01`
`1e 20 22 25 28 2b 30` = temperatures
`14 19 22 32 41 51 5f` = RPM percentages

I would like to play around with modifying https://github.com/audiohacked/OpenCorsairLink/blob/testing/protocol/asetek/fan.c#L125 and sending values I mentioned above. I just don't have enough knowledge about all of this because I spent like 20 minutes with the research so I am really hesitant with sending some bytes to the cooling of my CPU.

gipfelen, 2019-04-05:
Thanks to @metthal input I forked the project, and made the necessary changes to get the H100i Pro fan control fully working.

https://github.com/gipfelen/OpenCorsairLink

metthal, 2019-04-05:
That's great to hear! I haven't tried it myself yet but it's nice to know that it works. I'll definitely use it next time I want to use Linux on my machine.

metthal, 2019-04-05:
Also, another thing I just found out, in case you wanted your cooler to monitor the temperature from other sensor, it should be enough to read the other sensor temperature in software and send it to the cooler as `0xA5`. For example for 30 °C just send `a5 1e 00` every second.

metthal, 2019-04-05:
And if you want fixed RPM use `0x43` followed by fan channel (1 byte) and RPM (2 bytes). For fixed % use `0x42` followed by fan channel (1 byte) and % of max RPM (1 byte).

gipfelen, 2019-04-05:
Thank you for the information! :) Tomorrow I'll try to implement that.

gipfelen, 2019-04-06:
I implemented the RPM and PWM control, but I ignored the "write temperature" for now, since I cannot find a good entry point within the original project.

metthal, 2019-04-06:
That's great! Basically H100i Pro is now totally usable with your fork.

Yeah, the write temp would require OpenCorsairLink to run as some sort of a daemon.

closed 2019-10-02 by audiohacked