hyperkineticnerd/OpenCorsairLink #158: H150i fan speed problems

justdanyul, 2019-03-16:
I can't change the fan speeds of any of my fans on a H150i PRO. Regardless of which mode I pick, they all spin around 1000rpm.

Setting the pump works fine though, its just the fans that are an issue.

NoMod-Programming, 2019-03-17:
Hi, can you run OpenCorsairLink with the `--debug` flag? It's show more information that's useful for narrowing down the problem.

justdanyul, 2019-03-17:
Sure, the output is as follows

```
Checking USB device 0 (05dc:a205)...
Checking USB device 1 (1d6b:0003)...
Checking USB device 2 (1b1c:0c12)...
Corsair product detected. Checking if device is H150i Pro... Dev=0, CorsairLink Device Found: H150i Pro!
Checking USB device 3 (05e3:0610)...
Checking USB device 4 (046d:c062)...
Checking USB device 5 (05ac:0250)...
Checking USB device 6 (05e3:0610)...
Checking USB device 7 (0b05:18a3)...
Checking USB device 8 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H150i Pro
AA 12 34 01 00 04 00
Firmware: 1.0.4.0
hardware version returned: AB 12 34 02 00 00 00
A9 12 34 1F 00
Temperature 0: 31.00 C
00
41 12 34 00 03 1B
Fan 0:	Mode 0x00
	Current/Max Speed 795/0 RPM
00
41 12 34 01 00 00
Fan 1:	Mode 0x00
	Current/Max Speed 0/0 RPM
00
41 12 34 02 00 00
Fan 2:	Mode 0x00
	Current/Max Speed 0/0 RPM
function:corsairlink_asetekpro_pump_mode_read file: protocol/asetekpro/pump.c
pump mode response = 33 12 34 01
pump speed response = 31 12 34 08 52
Pump:	Mode 0x01 (AsetekProPumpBalanced)
	Current/Max Speed 2130/0 RPM
Setting fan to mode: 5
DEBUG: deinit done
```

please note, currently, only one FAN is connected, as I had to connect the others to my fan controller to, well, control them :-D

metthal, 2019-03-21:
This might be also related to issues with H100i (https://github.com/audiohacked/OpenCorsairLink/issues/157#issuecomment-474114932). There seems to be a new kind of protocol for changing fan speed which isn't implemented yet.

closed 2019-10-02 by audiohacked