hyperkineticnerd/OpenCorsairLink #159: H150i pro device_number = -1

Jason0409, 2019-03-19:
what does it mean? I try some command to control the fan but it doesn't work. I use --debug command, I get:
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (24c9:1003)...
Checking USB device 2 (04ca:0027)...
Checking USB device 3 (8087:0aaa)...
Checking USB device 4 (05e3:0608)...
Checking USB device 5 (048d:8297)...
Checking USB device 6 (05e3:0608)...
Checking USB device 7 (1d6b:0002)...
Checking USB device 8 (1d6b:0003)...
Checking USB device 9 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = -1

Jason0409, 2019-03-19:
I just checked the main.c, does device_number  = -1  means can't find the device?  whether I set up the cooler in a wrong way?

Jason0409, 2019-03-20:
after reerection, I got:
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (24c9:1003)...
Checking USB device 2 (04ca:0027)...
Checking USB device 3 (8087:0aaa)...
Checking USB device 4 (1b1c:0c12)...
Corsair product detected. Checking if device is H150i Pro... Dev=0, CorsairLink Device Found: H150i Pro!
Checking USB device 5 (05e3:0608)...
Checking USB device 6 (048d:8297)...
Checking USB device 7 (05e3:0608)...
Checking USB device 8 (1d6b:0002)...
Checking USB device 9 (1d6b:0003)...
Checking USB device 10 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = -1

device_number still -1

metthal, 2019-03-21:
Have you tried running it as root?

audiohacked, 2019-04-15:
`device_number` is an additional identifier used for the old/original coolit devices. For asetek, its not used. As for the H150i, It's not officially supported yet.

closed 2019-10-02 by audiohacked