hyperkineticnerd/OpenCorsairLink #160: Corsair One i160 Incorrect Stats / Firmware Displayed

Omarsf, 2019-03-26:
**Describe the bug**
Status indicates a device is detected but returns incorrect information "Firmware: 0.0.0.0
Temperature 0:  0.00 C".

**To Reproduce**
Steps to reproduce the behavior:
sudo ./OpenCorsairLink.elf --device 0

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Ubuntu Server 18.04.2 LTS

**Additional context**
Fresh Ubuntu installation on a corsair one i160
[image]

fan does seem operational but not sure how it will behave as temps increase
[image]

Thank you,

audiohacked, 2019-03-27:
Corsair One i160 isn't supported. It would be very hard for me to add support without owning one.

jpuderer, 2019-10-10:
Sigh... well, I own one, so I guess I'd better step up and work on this one.

I'll see if I can't figure this out.