hyperkineticnerd/OpenCorsairLink #163: Can see Commander Pro but no further details on OSX

shoikan, 2019-04-15:
I am running a hackintosh (OSX)/Windows 10 dual boot and have a Commander Pro connected, with 2x LL140 fans. Under Windows everything works as expected (obviously), but when booting into OSX the fans go to 100% which is quite a bit louder than I would like. I've compiled commit d0504cf331098981ef54f63de42c4fc38c6a386a.  The utility works (yay! didn't even expect that, tbh!) and it can see the Commander, but does not seem to be able to do anything with it:

```
Renes-iMac:OpenCorsairLink shoikan$ ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: Commander PRO!

Vendor: Corsair
Product: Commander PRO
Firmware: V0.0.0
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Temperature 3:  0.00 C
Output 12v:  0.00 V
Output 5v:  0.00 V
Output 3.3v:  0.00 V
Fan 0:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 1:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 2:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 3:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 4:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 5:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0```

Additionally, I can't seem to use any off the commands to control the fans.

Oh.. and thank you SO much for this utility!!

audiohacked, 2019-04-15:
It looks like you didn't run the utility as root or with sudo! Try that first.

shoikan, 2019-04-15:
> It looks like you didn't run the utility as root or with sudo! Try that first.

I did, but I posted the results from running it as a mere mortal. Apologies for not posting the right output:
```
Renes-iMac:OpenCorsairLink shoikan$ sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: Commander PRO!

Vendor: Corsair
Product: Commander PRO
Firmware: V0.0.0
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Temperature 3:  0.00 C
Output 12v:  0.00 V
Output 5v:  0.00 V
Output 3.3v:  0.00 V
Fan 0:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 1:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 2:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 3:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 4:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 5:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
```

audiohacked, 2019-04-15:
You might be running into an OS X quirk, the newer OS versions block access to hardware from user apps.

shoikan, 2019-04-15:
> You might be running into an OS X quirk, the newer OS versions block access to hardware from user apps.

Probably :) On the other hand, they do allow vendor specified drivers for hardware (thinking of a reverse-engineered driver for a flatbed scanner that was artificially obsoleted by Canon), so it should be possible to access things via the USB port. I will try and see if I can diagnose this at all from my end.

audiohacked, 2019-10-03:
@shoikan If you have any new information, please reopen this issue.

closed 2019-10-03 by audiohacked