hyperkineticnerd/OpenCorsairLink #164: Unable to set LED Temperature Mode on the Corsair H100i Pro

tyzbit, 2019-04-24:
**Describe the bug**
Trying to set temperature-controlled LEDs on H100i Pro results in an error "Unknown suboption `temp=24:27:30`"

**To Reproduce**
Steps to reproduce the behavior:
```
OpenCorsairLink.elf \
  --device 0 \
  --led mode=5,colors=00FF00:FFFF00:FF0000,temp=24:27:30
```

**Expected behavior**
The command completes successfully and the LED changes according to temperature

**Desktop (please complete the following information):**
 - OS: Linux Mint
 - Version 18.3

**Additional context**
On commit d0504cf.  I also tried `temps` but that doesn't work either.

drewhoener, 2019-04-24:
Use sudo? your steps without sudo produce the output, adding sudo results in it changing colors

tyzbit, 2019-04-24:
Here's the output from trying `sudo`:
```
$ sudo OpenCorsairLink.elf --device 0 --led mode=5,colors=00FF00:FFFF00:FF0000,temp=24:27:30
Unknown suboption `temp=24:27:30'                                                                                                                                                                                                            
Dev=0, CorsairLink Device Found: H100i Pro!                                                                                                                                                                                                  
                                                                                                                                                                                                                                             
Vendor: Corsair                                                                                                                                                                                                                              
Product: H100i Pro                                                                                                                                                                                                                           
Firmware: 1.0.4.0                                                                                                                                                                                                                            
Temperature 0: 26.06 C                                                                                                                                                                                                                       
Pump:   Mode 0x01 (AsetekProPumpBalanced)                                                                                                                                                                                                    
        Current/Max Speed 2100/0 RPM
```

audiohacked, 2019-04-25:
The option is called 'temperature', when I get the time I'll fix it to match the other suboptions (fan and pump)

tyzbit, 2019-04-25:
That fixed me right up, thank you!

audiohacked, 2019-07-01:
@tyzbit Did I fix this? If so, either close this issue or tell me to.

tyzbit, 2019-07-01:
If the option now matches the help text then this can be closed.

closed 2019-10-03 by audiohacked