hyperkineticnerd/OpenCorsairLink #167: Commander Mini

adzam5, 2019-05-04:
Even though you do not list support for the Commander Mini I figured I would try out OpenCorsairLink in hopes that the commands for the Commander Mini and the Commander Pro would be the same.

My testing environment is a KDE Neon (Ubuntu 18.04) VirtualBox VM running on a Windows 10 host.  The Commander Mini shows up on the host and guest as:

**Name:** Corsair Memory, Inc. Integrated USB Bridge [0200]  
**VID:** 1b1c  
**PID:** 0c04  

Attached to the Commander Mini is a H100i, two case fans and one LED strip.

Unfortunately, I was not able to control anything, but OpenCorsairLink did find a device and labeled it as "Whiptail".  I have no idea what that means, but I found it somewhat promising.

Below are the results of a `dump` and `debug`.  Let me know if there is any other information I can provide.

```console
$ sudo ./OpenCorsairLink.elf --dump 

Checking USB device 0 (1b1c:0c04)...
Corsair product detected. Checking if device is H80... No (device_id 0x3D)
Corsair product detected. Checking if device is Cooling Node... No (device_id 0x3D)
Corsair product detected. Checking if device is Lighting Node... No (device_id 0x3D)
Corsair product detected. Checking if device is H100... No (device_id 0x3D)
Corsair product detected. Checking if device is H80i... No (device_id 0x3D)
Corsair product detected. Checking if device is H100i... No (device_id 0x3D)
Corsair product detected. Checking if device is Whiptail... Dev=0, CorsairLink Device Found: Whiptail!
Checking USB device 1 (80ee:0021)...
Checking USB device 2 (1d6b:0001)...

DEBUG: scan done, start routines
DEBUG: device_number = -1
```

```console
$ sudo ./OpenCorsairLink.elf --debug

Checking USB device 0 (1b1c:0c04)...
Corsair product detected. Checking if device is H80... 
---- Packet dump: -----------------------------
81 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is Cooling Node... 
---- Packet dump: -----------------------------
82 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is Lighting Node... 
---- Packet dump: -----------------------------
83 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is H100... 
---- Packet dump: -----------------------------
84 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is H80i... 
---- Packet dump: -----------------------------
85 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is H100i... 
---- Packet dump: -----------------------------
86 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is Whiptail... 
---- Packet dump: -----------------------------
87 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
Dev=0, CorsairLink Device Found: Whiptail!
Checking USB device 1 (80ee:0021)...
Checking USB device 2 (1d6b:0001)...

DEBUG: scan done, start routines
DEBUG: device_number = -1
```

audiohacked, 2019-05-05:
Thank you for the report! I'll have to make a patch for the naming.

audiohacked, 2019-05-05:
Could you try `sudo ./OpenCorsairLink.elf --device 0 --debug` and post the output?

adzam5, 2019-05-05:
```console
$ sudo ./OpenCorsairLink.elf --device 0 --debug

Checking USB device 0 (1b1c:0c04)...
Corsair product detected. Checking if device is H80... 
---- Packet dump: -----------------------------
81 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is Cooling Node... 
---- Packet dump: -----------------------------
82 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is Lighting Node... 
---- Packet dump: -----------------------------
83 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is H100... 
---- Packet dump: -----------------------------
84 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is H80i... 
---- Packet dump: -----------------------------
85 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is H100i... 
---- Packet dump: -----------------------------
86 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
No (device_id 0x3D)
Corsair product detected. Checking if device is Whiptail... 
---- Packet dump: -----------------------------
87 07 3d 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
Dev=0, CorsairLink Device Found: Whiptail!
Checking USB device 1 (80ee:0021)...
Checking USB device 2 (1d6b:0001)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: Whiptail
Firmware: 1.1.6
Temperature 0: 175.75 C
Temperature 1: 175.75 C
Temperature 2: 175.75 C
Temperature 3: 175.75 C
Speed: 04 32
Max Speed: 08 0B
Fan 0:  Quiet Mode (4PIN)
        Current/Max Speed 1074/2059 RPM
Speed: 04 37
Max Speed: 08 15
Fan 1:  Quiet Mode (4PIN)
        Current/Max Speed 1079/2069 RPM
Speed: 00 00
Max Speed: 00 00
Fan 2:  Not connected or failed
        Current/Max Speed 0/0 RPM
Speed: 00 00
Max Speed: 00 00
Fan 3:  Not connected or failed
        Current/Max Speed 0/0 RPM
Speed: 00 00
Max Speed: 00 00
Fan 4:  Not connected or failed
        Current/Max Speed 0/0 RPM
Speed: 00 00
Max Speed: 00 00
Pump:   Mode 0x07
        Current/Max Speed 0/0 RPM
DEBUG: deinit done
```

adzam5, 2019-05-05:
In my original testing I tried controlling with `sudo ./OpenCorsairLink --device -1` (since that what it was reporting the device as in my original debug) and no `--device` at all.  However, I have now tried with `--device 0` and I do have some control!  I am able to change the color of the LED strip and set the modes on the two fans listed.  I am not able to set the fans to a fixed rpm or pwm, if I try either mode I get `Unsupported Fan Mode`.  Also, I am also not sure if changing the pump mode is doing actually anything.

adzam5, 2019-05-06:
The patch corrected the name!

Unfortunately, at this time OpenCorsairLink does not seem to be functional enough with the Commander Mini.  One thing you will notice in my previous `--debug` dump is the temperatures are not correct.  I figured this was due to my testing being within a virtual machine without access to the sensors.  However, I setup a dual boot to further test things and the temperatures are still reading 175.75.  I assume the temperature reading need to be correct in order for the fan speeds to automatically change.

Please let me know if there is any other information I can provide.

audiohacked, 2019-06-20:
@adzam5 Where do you have the Temperature probes connected to? I _hope_ that the only problem is the byte order for the temperatures.

audiohacked, 2019-10-03:
I'm closing this issue; I've added a few labels so when I get a commander mini I can come back and work on any bugs/fixes.

closed 2019-10-03 by audiohacked