hyperkineticnerd/OpenCorsairLink #172: Dump my devices

rbh00723, 2019-05-09:
Hi Guys (and Girls),

I am very new to using Ubuntu as my daily driver and found your project that I would like to contribute to.

So I compiled and installed the program and saw that it does not appear to have much in the way of mouse/keyboard/headset support I own a Dark Core mouse, K70 keyboard, and a Void wireless headset I want to perform a dump and upload the files so you guys may possibly write code for them. 

I am sorry that I can not contribute more in the way of code writing, I hope to one day be able to do such but today is not that day.

Anyway so I ran OpenCorsairLink.elf --dump however I am not sure where those files actually went nor do I know what I should do with them afterwards. Obviously I assume that I upload them here but not sure where exactly...

I appreciate what your doing here helping to make Linux more accessible to the main stream.

audiohacked, 2019-05-09:
The `sudo ./OpenCorsairLink.elf --dump` doesn't produce files, but prints info on Corsair Water Coolers and Power Supplies only. There are no plans to add mouse/keyboard/headset support as OpenCorsairLink mostly deals with water pumps, fans, and reading temperatures.

rbh00723, 2019-05-11:
That's ashame CKB seems lacking and I was hoping for someone to make something to help

audiohacked, 2019-06-19:
I'm sorry that CKB and OpenCorsairLink is unable to do what you want. It's **very** difficult to get any Corsair device supported. Usually for each device someone has to reverse engineer the protocol used to talk to the device, or be lucky enough that previous work can be applied to a new device.

closed 2019-10-03 by audiohacked