hyperkineticnerd/OpenCorsairLink #173: No output for AX1600i

lissyx, 2019-05-13:
**Describe the bug**
No output but device is found

**To Reproduce**
Steps to reproduce the behavior:
1. git clone repo
2. make -j
3. sudo ./OpenCorsairLink.elf

**Expected behavior**
I should see power usage and others like I did on HX1000i

**Desktop (please complete the following information):**
 - OS: Debian/Sid
 - Linux 4.9

**Additional context**
```
$ sudo ./OpenCorsairLink.elf --dump --device 0 --debug
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (046d:c51a)...
Checking USB device 2 (1d6b:0002)...
Checking USB device 3 (1d6b:0002)...
Checking USB device 4 (1d6b:0003)...
Checking USB device 5 (1d6b:0003)...
Checking USB device 6 (046d:c225)...
Checking USB device 7 (1b1c:1b2a)...
Checking USB device 8 (046d:c221)...
Checking USB device 9 (046d:c223)...
Checking USB device 10 (1d6b:0002)...
Checking USB device 11 (1d6b:0003)...
Checking USB device 12 (1d6b:0002)...
Checking USB device 13 (1d6b:0003)...
Checking USB device 14 (1d6b:0002)...
Checking USB device 15 (1d6b:0003)...
Checking USB device 16 (1b1c:1c11)...
Corsair product detected. Checking if device is AX1600i... Dev=0, CorsairLink Device Found: AX1600i!
Checking USB device 17 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
```

lissyx, 2019-05-13:
@audiohacked I see it's using a different driver / protocol than my `HX1000i`, referring to USB dongle rather than `RMI`. Is it just that nobody had time to reverse it, or is there some bigger issue ?

audiohacked, 2019-06-05:
@lissyx No one has tested or validated the HX1000i before. The PSU might be able to use the RMI driver, but the PSU is from the USB Dongle era and no one has reversed the USB dongle protocol yet, among other things.

lissyx, 2019-06-10:
>  No one has tested or validated the HX1000i before.

I guess you were to write AX1600i, because the HX did work. At least it did report values :)

> no one has reversed the USB dongle protocol yet, among other things

I don't know it qualifies and/or applied to the AX1600i, but trying to find something that works I landed on https://github.com/Hagbard-Celin/cpsumon/blob/master/libcpsumon/src/cpsumon.c that seems to have some bits.

However, I have been too lazy (lack of time, really) to go further, because using this required exposing a new VID/PID for the USB serial chip in the `cp210x` kernel driver.

Does this codebase seems to implement code for USB dongle or am I being mislead somehow ?

audiohacked, 2019-06-19:
I did a little research on the AX1600i. The AX1600i doesn't need the USB dongle protocol, that was for other AX PSUs that lacked integrated USB CorsairLink. Now, the AX1600i might have a different (unsupported) protocol or it may be supported using the RMi protocol. cpsumon doesn't look like it would support the AX1600i, since it only supports the AX760, AX860, AX1200, and AX1500. Note the lack of "i" on the model strings. If you didn't know, Corsair likes to use different protocols for each product and make changes via firmware update. So I don't know when support will arrive considering I don't own an AX1600i or any other AXi PSU.

lissyx, 2019-06-21:
> Now, the AX1600i might have a different (unsupported) protocol or it may be supported using the RMi protocol

I did hack into the codebase to try that, but failed. Would you have any suggestions on how I should proceed ?

> cpsumon doesn't look like it would support the AX1600i, since it only supports the AX760, AX860, AX1200, and AX1500. Note the lack of "i" on the model strings. If you didn't know, Corsair likes to use different protocols for each product and make changes via firmware update.

Nope, discovering the field, so I got tricked, thanks for the information :)

audiohacked, 2019-10-03:
Closing this as I do not have an AX1600i to test with. Sorry.

closed 2019-10-03 by audiohacked

lissyx, 2019-10-03:
> Closing this as I do not have an AX1600i to test with. Sorry.

Well, that's unfortunate. I still have mine, and I still have interest in supporting it, even though I lack time to reverse-engineer stuff from the Windows app.