hyperkineticnerd/OpenCorsairLink #176: H115i

Ryuuka9, 2019-06-22:
Hi, I have a corsair h115i, but in opencorsairlink it is recognized as h100i GT V2, why does this happen?

If I try to change the pump mode this is the result:
Dev=0, CorsairLink Device Found: H110i GT V2!

Vendor: Corsair
Product: H110i GT V2
Firmware: 2.8.0.0
Temperature 0: 35.00 C
Temperature 1: 35.00 C
Temperature 2: 35.00 C
Fan 0:	Mode 0x91
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
Pump:	Mode 0x03
	Current/Max Speed 2910/0 RPM
Setting fan to mode: 3
Setting pump to mode: 3
Unsupported Generic Pump Function

I also tried to put the fan mode on a custom curve, but even by setting the various temperatures and speeds, the fan speed is always high.

audiohacked, 2019-06-22:
The H115i is called H100i GT V2 internally to SIV and CorsairLink4.

audiohacked, 2019-06-22:
Duplicate of #152

closed 2019-06-22 by audiohacked