hyperkineticnerd/OpenCorsairLink #179: Temperature error on H150i

msmaldi, 2019-06-22:
**Describe the bug**
It is a simple bug in display temperature, after run 
`sudo ./OpenCorsairLink.elf --device 0`
`
...
Temperature 0: 36.05 C
...
`
but temperature is 36.5

**To Reproduce**
Steps to reproduce the behavior:
1. Start a application thats use all clocks 100%
2. Temperature go:
30.01
30.02
30.03
...
30.09
31.00

never exist 30.10, 30.11

Correct is
30.1
30.2
...
30.9

audiohacked, 2019-06-22:
How do you know that the temperature is suppose to be 36.5 when OCL prints 36.05?

msmaldi, 2019-06-22:
Why i temperature never between 30.10 and 30.99, 31.10 and 31.99
After 30,09 go direct to 31.00

msmaldi, 2019-06-22:
I think error is in file protocol/asetekpro/temperature
`    *( temperature ) = (double)response[3] + ( (double)response[4] / 100 );`
Correct
`    *( temperature ) = (double)response[3] + ( (double)response[4] / 10 );`

I have no time now to fix.

closed 2019-06-22 by audiohacked

jlask, 2019-07-12:
I had to revert this commit otherwise I was getting all zeroes for temp/fan output values.    

I also see, "Unable to detach kernel driver from Corsair".

I have the H100i Pro.