hyperkineticnerd/OpenCorsairLink #182: Lightning Node Pro Unable to claim USB device interface

Shaneee92, 2019-06-27:
When running the the application I get the **Unable to claim USB device interface**  error.

`Ryzen-Pro:~ shaneee$ sudo OpenCorsairLink.elf --dump
Checking USB device 0 (1b1c:0c0b)...
Corsair product detected. Checking if device is Lighting Node PRO... Unable to detach kernel driver from Corsair Device
Unable to claim USB device interface
DEBUG: scan done, start routines
DEBUG: device_number = -1`

Above is the dump from it. I’ve pulled and built from the latest source available. 

Running macOS Mojave 10.14.5 Hackintosh. Lightning Node Pro is shown in System Information under USB info. Any more info required I’ll be happy to provide it.

Shaneee92, 2019-07-01:
Should also note. I’ve just tried the feature/improved_scan branch and recieve this error instead,

`Corsair product detected. Checking if device is Lighting Node PRO... Corsair Device has no kernel driver attached`
`Encountered LibUSB Error: Access denied (insufficient permissions)!`

msmaldi, 2019-07-01:
I think have same problem

```
sudo ./OpenCorsairLink.elf --device 0 --dump
[sudo] password for msmaldi:    
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (8087:0aaa)...
Checking USB device 2 (1b1c:0c12)...
Corsair product detected. Checking if device is H150i Pro... Unable to detach kernel driver from Corsair Device
Dev=0, CorsairLink Device Found: H150i Pro!
Checking USB device 3 (05e3:0608)...
Checking USB device 4 (048d:8297)...
Checking USB device 5 (413c:8505)...
Checking USB device 6 (05e3:0608)...
Checking USB device 7 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H150i Pro
Firmware: 0.0.0.0
hardware version returned: 00 00 00 00 00 00 00
Temperature 0:  0.00 C
Fan 0:	Mode 0x00
	Current/Max Speed 0/0 RPM
Fan 1:	Mode 0x00
	Current/Max Speed 0/0 RPM
Fan 2:	Mode 0x00
	Current/Max Speed 0/0 RPM
function:corsairlink_asetekpro_pump_mode_read file: protocol/asetekpro/pump.c
pump mode response = 00 00 00 00
pump speed response = 00 00 00 00 00
Pump:	Mode 0x00 (AsetekProPumpQuiet)
	Current/Max Speed 0/0 RPM
DEBUG: deinit done
Unable to attach kernel driver to USB device
```

source83, 2019-07-05:
Same problem for me to:
`sudo ./OpenCorsairLink.elf --dump
[sudo] password for source: 
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (045e:0291)...
Checking USB device 2 (1bcf:2284)...
Checking USB device 3 (1d6b:0002)...
Checking USB device 4 (1d6b:0003)...
Checking USB device 5 (1b1c:0c0a)...
Corsair product detected. Checking if device is H110i GT V2... Unable to detach kernel driver from Corsair Device
Unable to claim USB device interface
DEBUG: scan done, start routines
DEBUG: device_number = -1
`

msmaldi, 2019-07-10:
@Shaneee92 source83

I fix in this problem with my H150i running back commit e8d79e6

`git reset --hard e8d79e6`

Shaneee92, 2019-07-11:
> @Shaneee92 source83
> 
> I fix in this problem with my H150i running back commit [e8d79e6](https://github.com/audiohacked/OpenCorsairLink/commit/e8d79e6d7d1173d7f59776660f9817c97c5ae96b)
> 
> `git reset --hard e8d79e6`

Now finds the device but no values etc.

```
Ryzen-Pro:~ shaneee$ sudo OpenCorsairLink.elf --device=0 --dump
Password:
Checking USB device 0 (0a12:0001)...
Checking USB device 1 (0781:5567)...
Checking USB device 2 (0d8c:0014)...
Checking USB device 3 (045e:0745)...
Checking USB device 4 (1b1c:0c0b)...
Corsair product detected. Checking if device is Lighting Node PRO... Dev=0, CorsairLink Device Found: Lighting Node PRO!

DEBUG: scan done, start routines
DEBUG: device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: Lighting Node PRO
Firmware: V0.0.0
DEBUG: string done
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Temperature 3:  0.00 C
Output 12v:  0.00 V
Output 5v:  0.00 V
Output 3.3v:  0.00 V
Fan 0:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 1:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 2:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 3:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 4:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 5:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Setting LED
```

audiohacked, 2019-10-03:
Closing this as I don't have hardware for testing or supporting it. Maybe someday support will be officially added.

closed 2019-10-03 by audiohacked