hyperkineticnerd/OpenCorsairLink #184: OpenCorsairLink vs Corsair Link (Windows)

audiohacked, 2019-07-02:
For anyone wondering, OpenCorsairLink sets/applies settings on a per-power-cycle basis; i.e. until next power off the settings are temporarily saved. I do know that some device protocols allow saving settings that survive a power cycle, but those protocol commands are undocumented and haven't been reversed engineered.

Also, OpenCorsairLink doesn't run as a service or daemon, which means temperature curves based on CPU temperature won't work. For now and until I decide to make a version of OpenCorsairLink that runs as a daemon, there won't be any solution to allow Linux users to do all features of the Windows version of Corsair Link.

DevolgaDevs, 2019-07-04:
Could be awesome !
Thank you for your hard work.

msmaldi, 2019-07-10:
I boot a Windows, and Corsair Link doesn't apply, how you make applied?
In my linux:
I move OpenCorsairLink.elf to /opt/OpenCorsairLink/OpenCorsairLink.elf
Create a file in /lib/systemd/system/corsairlink.service with content
```
[Unit]
Description=OpenCorsairLink service

[Service]
ExecStart=/opt/OpenCorsairLink/OpenCorsairLink.elf --device 0 --pump mode=3 --fan mode=3 --led channel=2,mode=5,colors=664400:ff3300:ff0000,temps=28:34:40

[Install]
WantedBy=multi-user.target
```
and for startup on boot
`sudo systemctl enable corsairlink.service`

works perfect after system boot, but before service load with a bad green color.