hyperkineticnerd/OpenCorsairLink #185: Commander PRO misdetected?

truatpasteurdotfr, 2019-07-05:
**Describe the bug**
my commander pro is detected as AX1200i

**To Reproduce**
Steps to reproduce the behavior:
1. build from source
2. sudo ./OpenCorsairLink.elf

**Expected behavior**
detect the "right" device

**Desktop (please complete the following information):**
 - OS: linux CentOS-7 x86-64

truatpasteurdotfr, 2019-07-05:
[truadm@ld18-1012 OpenCorsairLink]$ lsusb
...
Bus 001 Device 005: ID 1b1c:1d00 Corsair 
...

replacing the devide id as follows:
```
diff --git a/device.c b/device.c
index cfe8b87..c2538f7 100644
--- a/device.c
+++ b/device.c
@@ -242,7 +242,7 @@ struct corsair_device_info corsairlink_devices[] = {
     },
     {
         .vendor_id = 0x1b1c,
-        .product_id = 0x0c10,
+        .product_id = 0x1d00,
         .device_id = 0xFF,
         .name = "Commander PRO", /* Barbuda */
         .read_endpoint = 0x01 | LIBUSB_ENDPOINT_IN,
@@ -494,7 +494,7 @@ struct corsair_device_info corsairlink_devices[] = {
     },
     {
         .vendor_id = 0x1b1c,
-        .product_id = 0x1d00,
+        .product_id = 0x1c10,
         .device_id = 0xFF,
         .name = "AX1200i",
         .read_endpoint = 0x01 | LIBUSB_ENDPOINT_IN,

```
allows to read and set the fans PWM or RPM
```
[truadm@ld18-1012 OpenCorsairLink]$ sudo ./OpenCorsairLink.elf --dev 0
Dev=0, CorsairLink Device Found: Commander PRO!

Vendor: Corsair
Product: Commander PRO
Firmware: V0.8.210
Temperature 0: 34.39 C
Temperature 1: 33.74 C
Temperature 2: 31.91 C
Temperature 3: 31.48 C
Output 12v: 11.89 V
Output 5v:  4.88 V
Output 3.3v:  3.33 V
Fan 0:  Mode: Auto/Disconnected
        PWM: 50%
        RPM: 1138
Fan 1:  Mode: Auto/Disconnected
        PWM: 50%
        RPM: 1139
Fan 2:  Mode: Auto/Disconnected
        PWM: 50%
        RPM: 1080
Fan 3:  Mode: Auto/Disconnected
        PWM: 50%
        RPM: 1144
Fan 4:  Mode: Auto/Disconnected
        PWM: 50%
        RPM: 1100
Fan 5:  Mode: Auto/Disconnected
        PWM: 0%
        RPM: 0
```

I only miss the pump information/settings

truatpasteurdotfr, 2019-07-05:
btw, I had to manually build/install a newer libusb to make it work, as the system libusb is too old, but I did NOT replace the system version, I just side instal it and link the libs from /local/libusb/1.0.22/.

```
[truadm@ld18-1012 OpenCorsairLink]$ ldd OpenCorsairLink.elf
        linux-vdso.so.1 =>  (0x00007ffcdc5b7000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f6935960000)
        libusb-1.0.so.0 => /local/libusb/1.0.22/lib/libusb-1.0.so.0 (0x00007f6935747000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f693537a000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f6935c62000)
        libudev.so.1 => /lib64/libudev.so.1 (0x00007f6935164000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f6934f48000)
        librt.so.1 => /lib64/librt.so.1 (0x00007f6934d40000)
        libcap.so.2 => /lib64/libcap.so.2 (0x00007f6934b3b000)
        libdw.so.1 => /lib64/libdw.so.1 (0x00007f69348ec000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007f69346e8000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f69344d2000)
        libattr.so.1 => /lib64/libattr.so.1 (0x00007f69342cd000)
        libelf.so.1 => /lib64/libelf.so.1 (0x00007f69340b5000)
        libz.so.1 => /lib64/libz.so.1 (0x00007f6933e9f000)
        liblzma.so.5 => /lib64/liblzma.so.5 (0x00007f6933c79000)
        libbz2.so.1 => /lib64/libbz2.so.1 (0x00007f6933a69000)
```

audiohacked, 2019-10-02:
Closing this as I'm unsure of the actual problem.

closed 2019-10-02 by audiohacked