hyperkineticnerd/OpenCorsairLink #189: New CPU/Motherbord, H110i is no longer changing settings.

cbacott, 2019-07-29:
**Describe the bug**
Since upgrading to a Ryzen 3700 and Prime X370-pro, cannot control my Corsair H110i.
Command says it detects the device, and dmesg shows the scan successfully, however pump and fan speed do not change. Dump is showing something weird at the end:

```
$ sudo ./OpenCorsairLink.elf --dump
Checking USB device 0 (05e3:0616)...
Checking USB device 1 (1d6b:0003)...
Checking USB device 2 (05e3:0610)...
Checking USB device 3 (1d6b:0002)...
Checking USB device 4 (1d6b:0003)...
Checking USB device 5 (046d:c338)...
Checking USB device 6 (046d:c069)...
Checking USB device 7 (1d6b:0002)...
Checking USB device 8 (1d6b:0003)...
Checking USB device 9 (0a12:0001)...
Checking USB device 10 (1b1c:0c04)...
Corsair product detected. Checking if device is H80... No (device_id 0x42)
Corsair product detected. Checking if device is Cooling Node... No (device_id 0x42)
Corsair product detected. Checking if device is Lighting Node... No (device_id 0x42)
Corsair product detected. Checking if device is H100... No (device_id 0x42)
Corsair product detected. Checking if device is H80i... No (device_id 0x42)
Corsair product detected. Checking if device is H100i... No (device_id 0x42)
Corsair product detected. Checking if device is Commander Mini... No (device_id 0x42)
Corsair product detected. Checking if device is H100i GT... No (device_id 0x42)
Corsair product detected. Checking if device is H110i GT... No (device_id 0x42)
Corsair product detected. Checking if device is H110i... Dev=0, CorsairLink Device Found: H110i!
Checking USB device 11 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: device_number = -1
```

The following command used to set the first fan into full bore loud mode, it doesn't seem to do anything anymore.
```
$ sudo ./OpenCorsairLink.elf --fan channel=1,mode=5
Dev=0, CorsairLink Device Found: H110i!

```
**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04 (this is the same OS this used to work under)
- Version:
```
 $ sudo ./OpenCorsairLink.elf --version
OpenCorsairLink Version: v0.9.0.0-3a34Dev=0, CorsairLink Device Found: H110i!
```

 Compiled this from git, not showing any newer version available via git pull. I'm willing to dig into the code itself and adding debug statements if need be, just wondering if this is some kind of known problem before I start spending time on it. It looks like, from the dump command, that it's detecting it as device 0 correctly, but after that it's setting the device to -1.

mazunki, 2019-08-11:
I also need to add the `--device 0` option. I'm not being allowed to use fan `mode 0 or 1`. My fans were at 500rpm, now I got them up to 1000, unintentionally. Can't seem to lower them again.

cbacott, 2019-08-11:
Yeah, something changed for me. I'm now not able to run it with --device=0. I was looking at the code and decided to start fresh, so I rm'd the whole thing and re-cloned the git. Now, it won't do anything, if I set device=0 it says it's out of range. Not sure what changed here. I'm poking around the code.

audiohacked, 2019-08-13:
device_number is set to -1 so that it forces the user to select the intended device. What this hack does is always select the last detected device which breaks usability for myself (since I have several devices that uses this utility).

cbacott, 2019-08-14:
That is a bit embarrassing, I didn't consider that. This patch will allow selecting of another device, while also providing a default selection like before, and also spits out a good error message to notify the user that they need permission to access the devices. 

```
--- main.c.orig	2019-08-11 11:14:42.255704603 -0500
+++ main.c	2019-08-13 21:30:08.147805494 -0500
@@ -64,7 +64,7 @@
 {
     int rr; // result from libusb functions
 
-    int8_t device_number = -1;
+    int8_t device_number = 0;
 
     int scanlist_count = 0;
 
@@ -87,7 +87,7 @@
     msg_debug( "DEBUG: scan done, start routines\n" );
     msg_debug( "DEBUG: selected device_number = %d\n", device_number );
 
-    if ( device_number >= 0 )
+    if ( scanlist_count > 0 )
     {
         if ( device_number >= scanlist_count )
         {
@@ -119,6 +119,10 @@
             }
         }
     }
+    else
+    {
+        msg_info("No devices detected. You may not have permissions to access the device.\n");
+    }
 
 exit:
     corsairlink_close( context );
```

mazunki, 2019-08-14:
@cbacott 

My error is different. I made an issue for it at #195 .

mazunki, 2019-08-14:
Changing 
```-    int8_t device_number = -1;
+    int8_t device_number = 0;```

followed by `sudo make install`
does the trick for me, having only one device to care about.

closed 2019-10-02 by audiohacked