hyperkineticnerd/OpenCorsairLink #190: Asetek Pro (H150i Pro) libusb timeouts unless first "mounted" in Windows VM iCUE

rubbingalcoholic, 2019-08-03:
When running OpenCorsairLink on a fresh boot of Ubuntu 19.04 with H150i Pro (Asetek Pro driver), the program takes a very long time to run through each of the vendor / product / firmware / etc. readouts in the hydro_asetekpro.c file (approx one second each) and fails to read any information (returns all `00`s). It also fails to write settings like pump and fan modes.

I checked the libusb_bulk_transfer in the lowlevel/asetek.c file and these are getting timeout errors.

Interestingly, this problem stops happening after I "mount" the USB device in a Windows virtual machine running Corsair iCUE. I'm able to write settings changes in the virtual machine, but when I turn off the VM and go back to using OpenCorsairLink in Linux, the program is able to read and write settings perfectly until the next reboot.

My guess is there's some kind of initial protocol handshake that puts the Asetek Pro device in a "ready to receive commands" state that OpenCorsairLink is not sending, but I don't know enough about monitoring USB traffic to confirm this.

**To Reproduce**
Steps to reproduce the behavior:
1. `sudo /usr/local/bin/OpenCorsairLink.elf --device 0 --led channel=1,mode=0,colors=000000 --pump mode=5`
2. Asetek Pro stats readouts return all `00`s after timing out, settings write fails
3. Start Windows VM and assign the USB device to the VM
4. Open Corsair iCUE app and play with settings to your heart's content
5. Close the VM or reassign the USB device back to Linux
6. OpenCorsairLink works correctly until the next reboot.

audiohacked, 2019-08-07:
Please try to reproduce the behavior with master branch.

rubbingalcoholic, 2019-08-07:
Thanks man, it works on master. I got tripped up because the default branch for this repo was testing. Set this to run on boot and no more disco LEDs. You saved my sanity. Thanks for all the work you've done on this project, it's wonderful!

audiohacked, 2019-08-07:
Testing branch should be where you are running from. I'd like to figure out exactly what is making Master work but not Testing.

rubbingalcoholic, 2019-08-07:
I'll fork the repo and setup a pull request between the branches. Maybe it's something easy.

fttx, 2019-08-07:
I've the same issue on the master branch with macOS 10.14.6. Once i run the CUE software on a VM it starts working

audiohacked, 2019-08-13:
@rubbingalcoholic any progress?

rubbingalcoholic, 2019-08-13:
No, sorry. I'll try this week. I got caught up last week with some life issues but definitely want to help figure this out.

audiohacked, 2019-08-13:
No problem. I can understand, I've been busy with work myself.

rubbingalcoholic, 2019-08-16:
#196

rubbingalcoholic, 2019-08-16:
@fttx could you try the branch on my fork and let me know if that fixes on your MacOS setup?

fttx, 2019-08-16:
Yes, it works! 👍

closed 2019-10-02 by audiohacked