hyperkineticnerd/OpenCorsairLink #193: Unable to detach kernel driver from device

LukePeltier, 2019-08-12:
**Describe the bug**
When attempting to get the status of my H100i GT V2, I get nothing. In the dump, i get an error message saying "Unable to detach kernel driver from Corsair Device"

**To Reproduce**
1. run 
```
sudo OpenCorsairLink
```

**Expected behavior**
Should see status of device

**Desktop (please complete the following information):**
 - OS: Ubuntu 19.04
- Kernel: 5.0.0-23-generic

audiohacked, 2019-08-13:
Please post your output.

LukePeltier, 2019-08-16:
```
$ sudo OpenCorsairLink --dump
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (1b1c:1b13)...
Checking USB device 2 (1d6b:0002)...
Checking USB device 3 (2109:0812)...
Checking USB device 4 (2109:0812)...
Checking USB device 5 (1d6b:0003)...
Checking USB device 6 (046d:082d)...
Checking USB device 7 (1b1c:0c09)...
Corsair product detected. Checking if device is H100i GT V2... Unable to detach kernel driver from Corsair Device
Dev=0, CorsairLink Device Found: H100i GT V2!
Checking USB device 8 (1532:0067)...
Checking USB device 9 (2104:0127)...
Checking USB device 10 (b58e:9e84)...
Checking USB device 11 (2109:2812)...
Checking USB device 12 (2109:2812)...
Checking USB device 13 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: selected device_number = -1
Unable to attach kernel driver to USB device
```

LukePeltier, 2019-08-16:
```
$ sudo OpenCorsairLink 
Dev=0, CorsairLink Device Found: H100i GT V2!

$
```

atomobianco, 2019-08-20:
Hi,
same for me on Fedora 30 (Workstation Edition) Linux 5.2.8-200.fc30.x86_64
Compilation went fine, but giving same error as @LukePeltier 

```
$ sudo ./OpenCorsairLink.elf --device 0 --debug
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (1d6b:0002)...
Checking USB device 2 (1d6b:0003)...
Checking USB device 3 (0b05:185c)...
Checking USB device 4 (0c60:0007)...
Checking USB device 5 (05ac:0267)...
Checking USB device 6 (046d:c52b)...
Checking USB device 7 (0424:2514)...
Checking USB device 8 (046d:085c)...
Checking USB device 9 (1b1c:0c09)...
Corsair product detected. Checking if device is H100i GT V2... Unable to detach kernel driver from Corsair Device
Dev=0, CorsairLink Device Found: H100i GT V2!
Checking USB device 10 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: selected device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H100i GT V2
Firmware: 0.0.0.0
00 00
Temperature 0:  0.00 C
00 00
Temperature 1:  0.00 C
00 00
Temperature 2:  0.00 C
00
00 00
Fan 0:	Mode 0x00
	Current/Max Speed 0/0 RPM
Unsupported Generic Pump Function
00 00
Pump:	Mode 0x00
	Current/Max Speed 0/0 RPM
DEBUG: deinit done
Unable to attach kernel driver to USB device
```

LukePeltier, 2019-08-25:
What is this a duplicate of?

PR #199 (2019-09-02): Set libusb to automatically detach/attach the kernel driver
## Proposed changes

Both libusb_detach_kernel_driver and libusb_attach_kernel_driver are
perfectly fine to return errors if there was no kernel bound to the
device in the first place.

Instead of tracking that ourselves, just set libusb to do it automatically.

> When this is enabled libusb will automatically detach the kernel driver
> on an interface when claiming the interface, and attach it when
> releasing the interface.

libusb_set_auto_detach_kernel_driver will only fail if on that platform
libusb cannot detach a kernel driver  [... cut]
commit 99e1d72f (2019-09-02): Set libusb to automatically detach/attach the kernel driver (#199)

Both libusb_detach_kernel_driver and libusb_attach_kernel_driver are
perfectly fine to return errors if there was no kernel bound to the
device in the first place.

Instead of tracking that ourselves, just set libusb to do it automatically.

> When this is enabled libusb will automatically detach the kernel driver
> on an interface when claiming the interface, and attach it when
> releasing the interface.

libusb_set_auto_detach_kernel_driver will only fail if on that platform
libusb cannot detach a kernel driver in the first  [... cut]
closed 2019-09-02 by audiohacked