hyperkineticnerd/OpenCorsairLink #195: H100i not working as expected.

mazunki, 2019-08-14:
**Describe the bug**
Modes 0, 1, and 6 are not working. 

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the installation steps.

2.a) Try to run `sudo ./OpenCorsairLink.elf --device 0 --fan mode=6,temps=30:40:60,speeds=500:600:1000`

2.b) Try to run `sudo ./OpenCorsairLink.elf --device 0 --fan mode=1,rpm=500`
2.c) Try to run `sudo ./OpenCorsairLink.elf --device 0 --fan mode=0,pwm=25`

**Expected behavior**
I'm expecting modes 0 and 1 to set my RPM to around 500 (as 25% of 2162 is 540). I'm expecting mode 6 to set various thresholds. 

Instead, modes 0 and 1 are "unsupported", while mode 6 doesn't give any error message. Noting changes, either.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: `Linux mazunki-zeus 5.2.2-122.current #1 SMP PREEMPT Wed Jul 24 01:03:42 UTC 2019 x86_64 GNU/Linux`
```
NAME="Solus"
VERSION="4.0"
ID="solus"
VERSION_CODENAME=fortitude
VERSION_ID="4.0"
PRETTY_NAME="Solus 4.0 Fortitude"
ANSI_COLOR="1;34"
HOME_URL="https://getsol.us"
SUPPORT_URL="https://getsol.us/articles/contributing/getting-involved/en/"
BUG_REPORT_URL="https://dev.getsol.us/"
```

 - Version `OpenCorsairLink Version: v0.9.0.0-8ad6`

**Additional context**
I waited a bit between each command, so output is not affected by delays. When I configured my fans on Windows I had an RPM of around 500. This was in effect on Solus too, until I started messing with this script. Now I can't seem to return to it.

Mode 3 (quiet) gives me an RPM of ~1000, mode 4 (balanced) gives me an RPM of around ~1700, and mode 5 (performance) gives me an RPM of around ~2100. Mode 2 (default) is not supported. 

My temperature is around 33 degree Celsius.

audiohacked, 2019-08-14:
The print out from your commands are before you set the fan speeds.
Your H110i doesn't support setting RPM or PWM.

mazunki, 2019-08-14:
I could set fixed RPM speeds in Windows, so it should be possible. I don't know how, though. Why doesn't mode 6 work, though? The printouts don't change afterwards either. That's why I printed it twice.

malakudi, 2019-08-15:
Coolit/HID support is broken for modes 0,1 since https://github.com/audiohacked/OpenCorsairLink/commit/c66733fa601c74951c21bfa2501fb68d7b3dc7a1 and mode 2 also broke after https://github.com/audiohacked/OpenCorsairLink/commit/f2bb5b9af135c750871a1af1d83f556493f5d6cf
Mode 0,1 was tried to get implemented again with https://github.com/audiohacked/OpenCorsairLink/commit/cb42de5802588bc3055d42f04d371a348302cd25 but incorrectly at least for Coolit/HID (I only own H110i with Coolit/HID API support).
You can have support of modes 0,1,2 if you do the following:
git clone https://github.com/audiohacked/OpenCorsairLink.git
git checkout b910de4fa2c80a917ff1fef043c1438090170a01
then apply following patch
```
--- logic/options_fan.c	2019-08-15 18:13:26.485443316 +0300
+++ logic/options_fan.c	2019-08-15 18:13:51.661435214 +0300
@@ -52,12 +52,12 @@
             break;
 
         case SUBOPTION_FAN_PWM:
-            sscanf( value, "%hhu", &settings->speed_pwm );
+            sscanf( value, "%hhu", &settings->data );
             msg_debug( "FAN PWM = %s\n", value );
             break;
 
         case SUBOPTION_FAN_RPM:
-            sscanf( value, "%hu", &settings->speed_rpm );
+            sscanf( value, "%hu", &settings->data );
             msg_debug( "FAN RPM = %s\n", value );
             break;
 
```
and then run make
I am trying to fix the code for current git but haven't succeeded yet, the https://github.com/audiohacked/OpenCorsairLink/commit/f2bb5b9af135c750871a1af1d83f556493f5d6cf commit change many aspects of the code so the above patch is no longer enough. So for now, use the commit version I mentioned above, apply the patch, and 0,1,2 modes will work. Haven't checked yet for mode 6.

malakudi, 2019-08-15:
Mode 6 doesn't work for Coolit/HID, needs to be implemented correctly after all those code refactoring that have happened the last years. Coolit/HID needs 5 points of temp/rpm or temp/pwm values (need to look API, don't remember now). Will try to fix mode 6 too.

malakudi, 2019-08-15:
Issues that exist in current code for Coolit/HID:
1. Value of modes 0,1 is not shown, it is shown 0.
```
OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H110i!

Vendor: Corsair
Product: H110i
Firmware: 2.0.0
Temperature 0: 27.47 C
Fan 0:	Fixed RPM Mode (4PIN) set to 0
	Current/Max Speed 1124/1746 RPM
Fan 1:	Fixed RPM Mode (4PIN) set to 0
	Current/Max Speed 1106/1745 RPM
Pump:	Mode 0x85
	Current/Max Speed 2902/3300 RPM
```
while before commit https://github.com/audiohacked/OpenCorsairLink/commit/f2bb5b9af135c750871a1af1d83f556493f5d6cf is shown fine
```
Dev=0, CorsairLink Device Found: H110i!

Vendor: Corsair
Product: H110i
Firmware: 2.0.0
Temperature 0: 27.64 C
Fan 0:	Fixed RPM Mode (4PIN) set to 1100
	Current/Max Speed 1111/1746 RPM
Fan 1:	Fixed RPM Mode (4PIN) set to 1100
	Current/Max Speed 1088/1745 RPM
Pump:	Mode 0x85
	Current/Max Speed 2905/3300 RPM
```
2. Modes 0,1,2 cannot be set, works fine till commit https://github.com/audiohacked/OpenCorsairLink/commit/b910de4fa2c80a917ff1fef043c1438090170a01 and above mentioned patch
3. Mode 6 does not work, never worked for Coolit/HID since legacy branch

I will try to fix all those for current git, but till then use my above mentioned solution for Coolit/HID devices.

closed 2019-10-02 by audiohacked