hyperkineticnerd/OpenCorsairLink #198: OpenCorsairLink not found

olielvewen, 2019-08-25:
**Describe the bug**
I 'm noob with this software and I've followed the usage section with the first command. Same result with the second (help command).
I 've got an Corsair RM650i that I 've bought in July.  

**To Reproduce**
Steps to reproduce the behavior:
Just run an terminal and try to run OpenCorsairLink. I'm using the stable version and I've installed it with pacman.

**Expected behavior**
Getting something to work. Unfortunately, there are no docs about what you should get when you have an AIO or an alim like RM650i.

**Desktop (please complete the following information):**
 - OS: Manjaro 5.2.8-1-MANJARO x86_64
 - Browser [firefox]
 - Version [The last]]

**Additional context**
Here my terminal output :
`[olivier@thorielle ~]$ sudo ./OpenCorsairLink.elf
[sudo] Mot de passe de olivier : 
sudo: ./OpenCorsairLink.elf : commande introuvable
`

olielvewen, 2019-08-29:
I precise that I've installed OpenCorsairLink with pamac and it is the stable version (from AUR).
I have access at others version which are opencorsairlink-git, opencorsairlink-testing, opencorsairlink-testing-git.
Perhaps, it is not the good version. Which one to choose ?

jonasmalacofilho, 2019-08-29:
Try calling `OpenCorsairLink` instead of `OpenCorsairLink.elf`.

And AFAIK (but @audiohacked please correct me if I'm wrong) you will _generally_ want to use the 'testing' branch.  For this both `opencorsairlink-testing` and `opencorsairlink-testing-git` from AUR look fine.

audiohacked, 2019-08-30:
@jonasmalacofilho Yes, everyone will generally want to use the 'testing' branch. It will have all the latest bug fixes. The 'master' branch will be updated when *I* feel like testing branch is stable enough to be considered a full 'stable' release.

olielvewen, 2019-08-30:
After followed your advice, I have removed opencorsairlink. Then I have try opencorsairlink-testing-git and try after different combinations.

Here outputs : 
`[olivier@thorielle ~]$ sudo ./OpenCorsairLink
sudo: ./OpenCorsairLink : commande introuvable
[olivier@thorielle ~]$ sudo ./OpenCorsairLink.elf
sudo: ./OpenCorsairLink.elf : commande introuvable
[olivier@thorielle ~]$ sudo ./opencorsairlink
sudo: ./opencorsairlink : commande introuvable
[olivier@thorielle ~]$ sudo ./opencorsairlink-testing-git
sudo: ./opencorsairlink-testing-git : commande introuvable
[olivier@thorielle ~]$ sudo ./opencorsairlink-testing
sudo: ./opencorsairlink-testing : commande introuvable
[olivier@thorielle ~]$ sudo ./opencorsairlink-testing.elf
sudo: ./opencorsairlink-testing.elf : commande introuvable
[olivier@thorielle ~]$ sudo ./opencorsairlink-testing-git.elf
sudo: ./opencorsairlink-testing-git.elf : commande introuvable
[olivier@thorielle ~]$ sudo ./OpenCorsairLink.elf
sudo: ./OpenCorsairLink.elf : commande introuvable
[olivier@thorielle ~]$ sudo ./OpenCorsairLink-testing-git.elf
sudo: ./OpenCorsairLink-testing-git.elf : commande introuvable
`
I've checked if there was a binary and yes in usr/bin/.

I've tried something else and it worked. I must run OpenCorsairLink in root (obviously) without ./ and elf.

`[olivier@thorielle ~]$ OpenCorsairLink

[olivier@thorielle ~]$ ps -aux
==============================================================================
olivier   2282  0.5  0.7 2809108 248844 ?      Sl   15:08   1:06 /usr/lib/firefox/firefox -conten
root      2406  0.0  0.0      0     0 ?        I<   15:08   0:00 [kworker/2:2H]
olivier   2854  0.0  0.2 764408 93396 ?        Sl   15:10   0:04 crow
olivier   3518  0.0  0.6 2425752 227320 ?      Sl   15:12   0:04 qtcreator
olivier   3525  0.0  0.1 221004 58288 ?        S    15:12   0:00 /usr/lib/qtcreator/clangbackend 
root      4053  0.0  0.0      0     0 ?        I    15:17   0:00 [kworker/10:1-memcg_kmem_cache]
root      4084  0.0  0.0      0     0 ?        S    15:17   0:00 [scsi_eh_9]
root      4085  0.0  0.0      0     0 ?        I<   15:17   0:00 [scsi_tmf_9]
root      4086  0.0  0.0      0     0 ?        S    15:17   0:00 [usb-storage]
root      4097  0.0  0.0      0     0 ?        I    15:17   0:00 [kworker/4:0-events]
root      4128  0.0  0.0      0     0 ?        I    15:17   0:00 [kworker/1:1]
root      4171  0.0  0.0      0     0 ?        I    15:22   0:00 [kworker/8:0]
root      4416  0.0  0.0      0     0 ?        I<   15:33   0:00 [kworker/6:2H-radeon-crtc]
root      5453  0.0  0.0      0     0 ?        I    15:35   0:00 [kworker/14:0-cgroup_destroy]
root      5507  0.0  0.0      0     0 ?        I<   15:39   0:00 [kworker/6:1H]
root      5623  0.0  0.0      0     0 ?        I<   15:41   0:00 [kworker/14:2H]
root      6093  0.0  0.0      0     0 ?        I<   16:13   0:00 [kworker/4:2H]
olivier   6172  0.0  0.2 510528 75376 ?        Rl   16:18   0:03 /usr/bin/qterminal
olivier   6177  0.0  0.0   7560  4220 pts/0    Ss   16:18   0:00 /bin/bash
olivier   6200  0.0  0.4 657476 140988 ?       Sl   16:20   0:05 pamac-manager
root      6204  0.0  0.3 315428 101464 ?       Ssl  16:20   0:02 /usr/bin/pamac-system-daemon
olivier   6210  0.0  0.0 234368  6104 ?        Sl   16:20   0:00 /usr/lib/libproxy/pxgsettings or
root      7996  0.0  0.0      0     0 ?        I<   16:40   0:00 [kworker/12:0H]
olivier   8041  0.2  0.4 809472 157084 ?       Sl   16:44   0:17 nemo
root      8356  0.0  0.0      0     0 ?        I    17:43   0:00 [kworker/u32:1-events_power_effi
root      8521  0.0  0.0      0     0 ?        I    18:21   0:00 [kworker/u32:2-events_power_effi
root      8563  0.0  0.0      0     0 ?        I    18:35   0:00 [kworker/u32:0-events_unbound]
olivier   8601  0.0  0.0  13540  3592 pts/0    R+   18:36   0:00 ps -aux
[olivier@thorielle ~]$ sudo OpenCorsairLink
[sudo] Mot de passe de olivier : 
Dev=0, CorsairLink Device Found: RM650i!

[olivier@thorielle ~]$ 
==============================================================================
`
Now I must learn how to use it . ^_^
A first test to see if I can get an output and yes it worked.

`[olivier@thorielle ~]$ sudo OpenCorsairLink
[sudo] Mot de passe de olivier : 
Dev=0, CorsairLink Device Found: RM650i!

[olivier@thorielle ~]$ sudo OpenCorsairLink --dump --device 0 --debug
[sudo] Mot de passe de olivier : 
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (046d:c517)...
Checking USB device 2 (03f0:6411)...
Checking USB device 3 (1d6b:0002)...
Checking USB device 4 (1d6b:0003)...
Checking USB device 5 (1b1c:1b51)...
Checking USB device 6 (1038:12ad)...
Checking USB device 7 (1b1c:1c0a)...
Corsair product detected. Checking if device is RM650i... Dev=0, CorsairLink Device Found: RM650i!
Checking USB device 8 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: selected device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: CORSAIR
Product: RM650i
Firmware: NA
DEBUG: string done
03 8D C1 F0 00 00
Temperature 0: 49.65 C
03 8E 9C F0 00 00
Temperature 1: 40.18 C
Powered: 614442 (7d.  2h)
Uptime: 13242 (0d.  3h)
DEBUG: time done
03 88 CC F9 00 00
Supply Voltage: 230.00 V
03 EE 33 08 00 00
Total Watts: 104.00 W
DEBUG: supply done
Output 12v:
03 8B 08 D3 00 00
        Voltage 12.12 V
03 8C 1D F0 00 00
        Amps  7.50 A
03 96 2B 08 00 00
        Watts 88.00 W
Output 5v:
03 8B 3F D1 00 00
        Voltage  5.00 V
03 8C 2B E0 00 00
        Amps  2.75 A
03 96 1B F8 00 00
        Watts 14.00 W
Output 3.3v:
03 8B D5 D0 00 00
        Voltage  3.34 V
03 8C 1C E0 00 00
        Amps  1.75 A
03 96 0B F8 00 00
        Watts  6.00 W
[olivier@thorielle ~]$ 
`

olielvewen, 2019-08-30:
Did am I alone in this case for run OpenCorsairLink ?i.e sudo OpenCorsairLink instead of ./OpenCorsairLink.elf
Feel free to close it if you think so. 
Thanks to Jonas for point me in the right direction. More explanation on the use of branches and the operation of the software would be welcome

closed 2019-08-31 by olielvewen