hyperkineticnerd/OpenCorsairLink #200: CLI documentation and behavior

jonasmalacofilho, 2019-08-30:
The README has a small mistake on the usage example.

```
$ sudo ./OpenCorsairLink.elf --device 0 --led channel=0,mode=0,color=00FF00
```

`color` should be `colors`:

```
$ sudo ./OpenCorsairLink.elf --device 0 --led channel=0,mode=0,colors=00FF00
```

---

Additionally, when there's only one device but `--device` is not passed the behavior is not very clear:

```
$ sudo ./OpenCorsairLink.elf --led channel=0,mode=0,colors=00FF00      
Dev=0, CorsairLink Device Found: Asetek 690LC!
```

The program exits without errors but no changes are made.  Only `--debug` shows a subtle indication that something might be wrong:

```
$ sudo ./OpenCorsairLink.elf --led channel=0,mode=0,colors=00FF00      
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (1d6b:0002)...
Checking USB device 2 (1d6b:0003)...
Checking USB device 3 (1e71:170e)...
Checking USB device 4 (046d:c08c)...
Checking USB device 5 (04d9:0169)...
Checking USB device 6 (046d:0a0e)...
Checking USB device 7 (05e3:0607)...
Checking USB device 8 (2433:b200)...
Corsair product detected. Checking if device is Asetek 690LC... Dev=0, CorsairLink Device Found: Asetek 690LC!
Checking USB device 9 (1e71:1714)...
Checking USB device 10 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: selected device_number = -1
```

(note: this is a dummy device I'm using to test OpenCorsairLink locally)

audiohacked, 2019-09-02:
@jonasmalacofilho what do you use as/for your dummy device?

jonasmalacofilho, 2019-09-02:
EVGA CLC, specially the 120: it's a newer Asetek (non Pro) model.

A few changes have been made to the protocol (exclusive command for reading the status; new lighting mode; the pump is sent a "level" instead of the plain duty value; maybe one other thing), but it should still respond if treated like a H80i GT.

P.S. If you want to take a look, [here's my notes on it](https://github.com/jonasmalacofilho/liquidctl/blob/master/docs/development/asetek690lc/evga-clc-capture-1.txt).

audiohacked, 2019-12-11:
Fix has been merged into testing.

closed 2019-12-11 by audiohacked