hyperkineticnerd/OpenCorsairLink #201: Unsupported Generic Fan Function

scott-hyperdrive, 2019-09-04:
**Describe the bug**
Attempting to set Modes (2 default, 3 quiet, 4 balanced, 5 Performance, 6 Custom) Either errors with 'Unsupported Generic Fan Function', sets all fans to  PWM: 0%, RPM: 0 or has no effect at all.

**To Reproduce**
Steps to reproduce the behavior:
1. Issue appropriate command (see examples below)
2. Review results of command
3. See error in cases where error is reported
4. Confirm RPM/PWM Speeds/%

**Expected behavior**
Mode to be set or curve to be set after issuing proper command

**Desktop (please complete the following information):**
Linux - Gentoo
OpenCorsairLink Version: v0.9.0.0

**Additional context**
First let me open by saying that I love the app and the idea behind it, it's just not working in my particular use case.

Trying to set any of the modes for the Commander Pro report as "Unsupported Generic Fan Function". I have tried using the '--fan channel=' options (OpenCorsairLink.elf --dev=0 --fan=_x_,mode=_x_) which returns the 'Unsupported...' response. I have also tried 'global' (sudo OpenCorsairLink.elf --dev=0 mode=_x_) which reports the same error. I can change fan speed with both RPM and PWM % as well as changing the mode for the H150i Pro (--dev=1 --pump mode=_x_). Lastly, I have tried creating a custom curve using both PWM mode and RPM mode and although I don't get any error, it just doesn't change the fan speeds from previous manually set state.

Example: 
```
Before Command with temp @ 40c:
Fan 1:  Mode: 4-Pin
        PWM: 65%
        RPM: 983

~ $ sudo OpenCorsairLink.elf --dev=0 --fan channel=1,mode=6,temps=55:65:75,speeds=55:75:100
Fan 1:  Mode: 4-Pin
        PWM: 65%
        RPM: 1069
```

The problem that I'm facing is that now that I have OpenCorsairLink installed is that the fan speed doesn't change at all anymore. On boot, it has all fans at 750rpm and reports PWM percentage as 0% (all fans are 4pin PWM). I have it set to one of the presets in Windows in hopes that it would be stored in the on board memory, but it doesn't/doesn't stay applied in Linux. So I have to manually change the fanspeeds by watching the temps and adjusting the speeds accordingly. Is there a way to completely uninstall OpenCorsairLink in hopes that the preset from Windows will work in Linux. Or do you have any suggestions on how to get the modes to apply to the fans? Really stuck at this point.

Please let me know if there is any additional information I can provide. I'm more than willing to provide whatever you need.

```
$ sudo OpenCorsairLink.elf 
Dev=0, CorsairLink Device Found: Commander PRO!
Dev=1, CorsairLink Device Found: H150i Pro!
Dev=2, CorsairLink Device Found: USB Dongle!
```

Current setting after manually setting all fans to 75% PWM

```
Dev=0, CorsairLink Device Found: Commander PRO!
Dev=1, CorsairLink Device Found: H150i Pro!
Dev=2, CorsairLink Device Found: USB Dongle!

Vendor: Corsair
Product: Commander PRO
Firmware: V0.8.210
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Temperature 3:  0.00 C
Output 12v: 11.87 V
Output 5v:  4.98 V
Output 3.3v:  3.33 V
Fan 0:  Mode: Auto/Disconnected
        PWM: 75%
        RPM: 1177
Fan 1:  Mode: 4-Pin
        PWM: 75%
        RPM: 1220
Fan 2:  Mode: 4-Pin
        PWM: 75%
        RPM: 1226
Fan 3:  Mode: 4-Pin
        PWM: 75%
        RPM: 1209
Fan 4:  Mode: 4-Pin
        PWM: 75%
        RPM: 1188
Fan 5:  Mode: 4-Pin
        PWM: 75%
        RPM: 1216
```

After issuing mode= command to fan/channel 0:

```
OpenCorsairLink.elf --dev=0 --fan channel=0,mode=3

Dev=0, CorsairLink Device Found: Commander PRO!
Dev=1, CorsairLink Device Found: H150i Pro!
Dev=2, CorsairLink Device Found: USB Dongle!

Vendor: Corsair
Product: Commander PRO
Firmware: V0.8.210
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Temperature 3:  0.00 C
Output 12v: 11.83 V
Output 5v:  5.00 V
Output 3.3v:  3.34 V
Fan 0:  Mode: Auto/Disconnected
        PWM: 0%
        RPM: 0
Fan 1:  Mode: 4-Pin
        PWM: 100%
        RPM: 1576
Fan 2:  Mode: 4-Pin
        PWM: 100%
        RPM: 1574
Fan 3:  Mode: 4-Pin
        PWM: 100%
        RPM: 1564
Fan 4:  Mode: 4-Pin
        PWM: 100%
        RPM: 1560
Fan 5:  Mode: 4-Pin
        PWM: 100%
        RPM: 1562
Unsupported Generic Fan Function
```
I have tried many other combinations of commands to test and make sure that I'm not missing something from the help section. I can supply every command attempt and the results if that is of help.

Sorry for the lengthy submission, just trying to give as much info as I can to see if there's some solution to this issue.

scott-hyperdrive, 2019-10-16:
Just bumping to see if there are any ideas/things for me to try to resolve the above problem.

audiohacked, 2019-10-17:
At this moment the Commander Pro only supports fixed RPM or fixed PWM. Any other modes aren't directly supported by the hardware. Or, the command for the other modes aren't known.

closed 2019-10-17 by audiohacked