hyperkineticnerd/OpenCorsairLink #202: Add support for changing mode for all fans and pump mode in one line

cryptomilk, 2019-10-01:
Hi,

I've looked into packaging OpenCorsairLink for Fedora. It would be possible to create a systemd service to execute it during boot to set the fan, led and pump modes. To achieve that I need to be able to execute OpenCorsairLink only once to set all the modes.

Example:

`OpenCorsairLink --device 0 --fan channel=0,mode=4 --fan channel=1,mode=4 --fan channel=2,mode=4 --pump mode=3 --led channel=0,mode=5`

Could you please add support for this?

Great that you use cmocka! ;-)

cryptomilk, 2019-10-01:
Or a config file would be another option :-)

darix, 2019-10-01:
Maybe with a simple format like this?

```yaml
devices:
- pump:
    mode: 3
  fan:
  - channel: 0
    mode: 4
  - channel: 1
    mode: 4
  - channel: 2
    mode: 44
  led:
  - channel: 0
    mode: 5
```

audiohacked, 2019-10-02:
Systemd already supports a OneShot execution and also using multiple commands for starting a service.

darix, 2019-10-02:
that is true but kinda ugly for a distro integration. do you really 
think offering the average user to edit service files is a good final 
solution?

audiohacked, 2019-10-02:
Since most, if not all, init systems support executing multiple commands for starting a service. This request is out-of-scope for OpenCorsairLink.

closed 2019-10-02 by audiohacked