hyperkineticnerd/OpenCorsairLink #211: [MacOS] Error Platform does not support kernel detachment

phaus, 2019-11-19:
**Describe the bug**
After the upgrade to macOS Catalina (10.15.2), I am not able to run the Program anymore.
It could also be an error with libusb.

**To Reproduce**
Steps to reproduce the behavior:
1. Update MacOS to the latest DEV Version
2. Run `sudo ./OpenCorsairLink.elf --dump`
```
Checking USB device 0 (046d:c52b)...
Checking USB device 1 (1b1c:0c08)...
Corsair product detected. Checking if device is H80i GT V2... Platform does not support kernel detachment
Dev=0, CorsairLink Device Found: H80i GT V2!
Checking USB device 2 (1b1c:0c04)...
Corsair product detected. Checking if device is H80... Platform does not support kernel detachment
Unable to claim USB device interface
DEBUG: scan done, start routines
DEBUG: selected device_number = -1
```
3. Run `sudo ./OpenCorsairLink.elf --device 0`
```
Platform does not support kernel detachment
Dev=0, CorsairLink Device Found: H80i GT V2!
Platform does not support kernel detachment
Unable to claim USB device interface
Detected 0 device(s), submitted device 0 is out of range
```

**Expected behavior**
See a listing of the device information.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser hm Firefox
 - Version Catalina 10.15.2

jonasmalacofilho, 2019-11-20:
The issue is not with detaching the kernel driver (that's indeed expected on MacOS), but with a failure to claim the interface.

Maybe there's something useful in the LIBUSB debug output?

```
sudo LIBUSB_DEBUG=4 ./OpenCorsairLink.elf --device 0
```

---

I should have used `msg_debug` instead of `msg_error` in https://github.com/audiohacked/OpenCorsairLink/blob/110977dac7ee3aa9d60d8a3f3e8de203164df988/logic/scan.c#L105-L109

For reference `libusb_set_auto_detach_kernel_driver` [returns](http://libusb.sourceforge.net/api-1.0/group__dev.html#ga1656d6b7272d4c82f842b20a834b1467):

> LIBUSB_SUCCESS on success  
> LIBUSB_ERROR_NOT_SUPPORTED on platforms where the functionality is not available

phaus, 2019-11-20:
FYI: Installed is a `H80i v2` and one `CORSAIR Commander Mini`.
I just run the cmd with debug enabled:

```
% sudo LIBUSB_DEBUG=4 ./OpenCorsairLink.elf --device 0
Password:
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000043] [00000307] libusb: debug [libusb_init] created default context
[ 0.000121] [00000307] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.000444] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb9084670
[ 0.000479] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb9084670
[ 0.005765] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.005783] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.005787] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.005791] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.005795] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.005798] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.005801] [00000307] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.005805] [00000307] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x8087
[ 0.005810] [00000307] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8002
[ 0.005813] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0005
[ 0.005816] [00000307] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.005820] [00000307] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.005823] [00000307] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.005826] [00000307] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.005862] [00000307] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.005873] [00000307] libusb: debug [process_new_device] allocating new device in context 0x7ffd2a500000 for with session 0xb9084670
[ 0.005884] [00000307] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x7ffd2a50265a
[ 0.006001] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xb90831ee
[ 0.006030] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xb90831ee/0x1a100000 against cached device with sessionID/locationID 0xb9084670/0x1d100000
[ 0.006036] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xb90831ee
[ 0.006510] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.006525] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.006529] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.006533] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.006536] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.006539] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.006543] [00000307] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.006546] [00000307] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x8087
[ 0.006550] [00000307] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x800a
[ 0.006553] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0005
[ 0.006557] [00000307] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.006560] [00000307] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x00
[ 0.006563] [00000307] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.006566] [00000307] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.006598] [00000307] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.006603] [00000307] libusb: debug [process_new_device] allocating new device in context 0x7ffd2a500000 for with session 0xb90831ee
[ 0.006609] [00000307] libusb: debug [process_new_device] found device with address 1 port = 1 parent = 0x0 at 0x7ffd2a5029fa
[ 0.006717] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0xe007cde3
[ 0.006745] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xe007cde3/0x1a120000 against cached device with sessionID/locationID 0xb90831ee/0x1a100000
[ 0.006751] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0xe007cde3/0x1a120000 against cached device with sessionID/locationID 0xb9084670/0x1d100000
[ 0.006756] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0xe007cde3
[ 0.007175] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.007190] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.007197] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[... cut ...]
[ 0.011548] [00000307] libusb: debug [libusb_release_interface] interface 0
[ 0.011790] [00000307] libusb: debug [libusb_close]
[ 0.011895] [00000307] libusb: debug [libusb_exit]
[ 0.011909] [00000307] libusb: debug [libusb_exit] destroying default context
[ 0.011915] [00000307] libusb: warning [libusb_exit] application left some devices open
[ 0.011918] [00000307] libusb: debug [usbi_remove_pollfd] remove fd 3
[ 0.012029] [0000120b] libusb: debug [darwin_event_thread_main] darwin event thread exiting
```

jonasmalacofilho, 2019-11-20:
This is the relevant line:

```
[ 0.011518] [00000307] libusb: error [darwin_claim_interface] USBInterfaceOpen: another process has device opened for exclusive access
```

Does any specific software come to mind as the possible culprit?

phaus, 2019-11-21:
Hi. Thank you for the quick response. 
I couldn't any any program or kernel module, that uses that USB Device.

I found a thread at stackoverflow about the same issue:

https://stackoverflow.com/questions/20253350/unable-to-claim-usb-interface-with-c-libusb-on-mac-os-x

Unfortunetally, the kext that was mentioned there, is not present in catalina anymore.

jonasmalacofilho, 2019-11-22:
I looked over the logs again and noticed that the issue is with the `1b1c:0c04` device, which should be your Commander Mini.

I'm not the maintainer here, but IIRC these `pid == 0x0c04` devices are HIDs, which would automatically be bound to a default HID driver/kext, thus explaining the problem.

Can you check whether the Commander Mini is indeed a HID, and also whether there is kext installed (is this the right lingo on macOS?) for it?

phaus, 2019-12-06:
I will try to do some tests when I have access to the system again. 
But I have to find out how to check which kext is using the device :-).

aloserini, 2020-01-17:
have the same problem:
USBInterfaceOpen: another process has device opened for exclusive access
any help?

luxus, 2020-03-04:
`➜  OpenCorsairLink git:(testing) sudo ./OpenCorsairLink.elf --device 0
Platform does not support kernel detachment
Dev=0, CorsairLink Device Found: H150i Pro!
Platform does not support kernel detachment
Unable to claim USB device interface
Detected 0 device(s), submitted device 0 is out of range
➜  OpenCorsairLink git:(testing) sudo LIBUSB_DEBUG=4 ./OpenCorsairLink.elf --device 0
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000008] [00000307] libusb: debug [libusb_init] created default context
[ 0.000032] [00000307] libusb: debug [libusb_init] libusb v1.0.23.11397
[ 0.000135] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1a2f959d7c3
[ 0.000145] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1a2f959d7c3
[ 0.001491] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.001496] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.001497] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.001498] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0xef
[ 0.001499] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x02
[ 0.001500] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.001501] [00000307] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.001502] [00000307] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x1235
[ 0.001503] [00000307] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x8202
[ 0.001504] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x041b
[ 0.001505] [00000307] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x01
[ 0.001533] [00000307] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x03
[ 0.001538] [00000307] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.001539] [00000307] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x02
[ 0.001662] [00000307] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.001669] [00000307] libusb: debug [process_new_device] allocating new device in context 0x7fcb3dc07100 for with session 0x1a2f959d7c3
[ 0.001679] [00000307] libusb: debug [process_new_device] found device with address 1 port = 4 parent = 0x0 at 0x7fcb3dd04e0a
[ 0.001725] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1a2fdea7d1a
[ 0.001734] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x1a2fdea7d1a/0x14900000 against cached device with sessionID/locationID 0x1a2f959d7c3/0x14200000
[ 0.001736] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1a2fdea7d1a
[ 0.002006] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[ 0.002012] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDescriptorType:    0x01
[ 0.002013] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdUSB:             0x0200
[ 0.002014] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceClass:       0x09
[ 0.002015] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceSubClass:    0x00
[ 0.002016] [00000307] libusb: debug [darwin_cache_device_descriptor]   bDeviceProtocol:    0x01
[ 0.002017] [00000307] libusb: debug [darwin_cache_device_descriptor]   bMaxPacketSize0:    0x40
[ 0.002018] [00000307] libusb: debug [darwin_cache_device_descriptor]   idVendor:           0x1a40
[ 0.002019] [00000307] libusb: debug [darwin_cache_device_descriptor]   idProduct:          0x0101
[ 0.002020] [00000307] libusb: debug [darwin_cache_device_descriptor]   bcdDevice:          0x0111
[ 0.002021] [00000307] libusb: debug [darwin_cache_device_descriptor]   iManufacturer:      0x00
[ 0.002043] [00000307] libusb: debug [darwin_cache_device_descriptor]   iProduct:           0x01
[ 0.002048] [00000307] libusb: debug [darwin_cache_device_descriptor]   iSerialNumber:      0x00
[ 0.002049] [00000307] libusb: debug [darwin_cache_device_descriptor]   bNumConfigurations: 0x01
[ 0.002064] [00000307] libusb: debug [darwin_check_configuration] active config: 1, first config: 1
[ 0.002066] [00000307] libusb: debug [process_new_device] allocating new device in context 0x7fcb3dc07100 for with session 0x1a2fdea7d1a
[ 0.002075] [00000307] libusb: debug [process_new_device] found device with address 2 port = 12 parent = 0x0 at 0x7fcb3de040fa
[ 0.002140] [00000307] libusb: debug [darwin_get_cached_device] finding cached device for sessionID 0x1a3027b03ea
[ 0.002152] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x1a3027b03ea/0x14800000 against cached device with sessionID/locationID 0x1a2fdea7d1a/0x14900000
[ 0.002154] [00000307] libusb: debug [darwin_get_cached_device] matching sessionID/locationID 0x1a3027b03ea/0x14800000 against cached device with sessionID/locationID 0x1a2f959d7c3/0x14200000
[ 0.002156] [00000307] libusb: debug [darwin_get_cached_device] caching new device with sessionID 0x1a3027b03ea
[ 0.002319] [00000307] libusb: debug [darwin_cache_device_descriptor] cached device descriptor:
[... cut ...]
[ 0.005310] [00000307] libusb: debug [libusb_release_interface] interface 0
[ 0.005387] [00000307] libusb: debug [libusb_close]
[ 0.005420] [00000307] libusb: debug [libusb_exit]
[ 0.005423] [00000307] libusb: debug [libusb_exit] destroying default context
[ 0.005424] [00000307] libusb: warning [libusb_exit] application left some devices open
[ 0.005425] [00000307] libusb: debug [usbi_remove_pollfd] remove fd 3
[ 0.005460] [0000090b] libusb: debug [darwin_event_thread_main] darwin event thread exiting`

I think I have the same issue here..