hyperkineticnerd/OpenCorsairLink #217: Unsupported Generic Pump Function for H80i GT V2

JoshDi, 2019-12-19:
**Describe the bug**
When trying to change the pump speed on the Corsair H80i GT V2, I get the error "Unsupported Generic Pump Function"

I am running the latest testing branch. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run: OpenCorsairLink.elf --device=0 --pump mode=5

Dev=0, CorsairLink Device Found: H80i GT V2!

Vendor: Corsair
Product: H80i GT V2
Firmware: 2.8.0.0
Temperature 0: 43.20 C
Temperature 1: 43.20 C
Temperature 2: 43.20 C
Fan 0:  Mode 0x30
        Current/Max Speed 0/0 RPM
**Unsupported Generic Pump Function**
Pump:   Mode 0x00
        Current/Max Speed 2700/0 RPM

**Expected behavior**
I'd like to be able to change my pump speed via OpenCorsairLink

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.03 LTS
 - Version: Latest testing branch of OpenCorsairLink (12ef1b6)

audiohacked, 2019-12-22:
This one should be pretty easy to fix. I think there's a disconnect between the logic for the Asetek and the supported functions for the Asetek driver.

drammelt, 2020-01-27:
Can confirm that fixes this issue

```
Dev=0, CorsairLink Device Found: H80i GT V2!

Vendor: Corsair
Product: H80i GT V2
Firmware: 2.8.0.0
Temperature 0: 33.60 C
Temperature 1: 33.60 C
Temperature 2: 33.60 C
Fan 0:	Mode 0x7C
	Current/Max Speed 0/0 RPM
Pump:	Mode 0x00
	Current/Max Speed 1050/0 RPM
```

commit 8a1e7fd5 (2019-12-23): Fix Asetek Pump

Fixes #217
commit c096f5aa (2019-12-23): Fix Asetek Pump

Fixes #217
PR #227 (2020-01-29): Fix Asetek Pump
Fixes #217

## Proposed changes

Describe the big picture of your changes here to communicate to the maintainers why we should accept this pull request. If it fixes a bug or resolves a feature request, be sure to link to that issue.

## Types of changes

What types of changes does your code introduce to OpenCorsairLink?
_Put an `x` in the boxes that apply_

- [ ] Bugfix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
 [... cut]
commit 46dbf206 (2020-01-29): Fix Asetek Pump

Fixes #217
closed 2020-01-29 by audiohacked