hyperkineticnerd/OpenCorsairLink #220: Add support for the Corsair One i164 system

raw-bin, 2019-12-28:
My impression is that this system is currently unsupported.

However, attempting to use OCL does produce some output. It is not clear to me whether that output is correct and/or complete.

Some relevant details:

High level system specs are [here.](https://www.corsair.com/uk/en/Categories/Products/Systems/CORSAIR-ONE/CORSAIR-ONE-GAMING-PC/p/CS-9020007-UK#tab-tech-specs)

Misc hw probe info:

```
$ lsusb -d 1b1c: -v

Bus 001 Device 003: ID 1b1c:0c14 Corsair Corsair ONE
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c14
  bcdDevice            1.00
  iManufacturer           1
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
[... 655 lines cut ...]
        Current/Max Speed 0/0 RPM
00
00 00 00 00 00 00
Bad Response
Fan 1:  Mode 0x00
        Current/Max Speed 0/0 RPM
function:corsairlink_asetekpro_pump_mode_read file: protocol/asetekpro/pump.c
pump mode response = 00 00 00 00
Bad Response for astekpro pump mode
pump speed response = 00 00 00 00 00
Bad Response for astekpro pump speed
Pump:   Mode 0x00 (AsetekProPumpQuiet)
        Current/Max Speed 0/0 RPM
Setting fan to mode: 0
42 00 32
DEBUG: deinit done
```

I get the feeling that this system is a variant of H110i Pro.

My primary need is fan and pump control which I hope to then plumb into upowerd etc.

My knowledge in this space is rudimentary but I am happy to dig as needed given pointers.

Thanks!

raw-bin, 2020-01-01:
@audiohacked : By any chance, is there a knowledge repository that could help n00bs like me to investigate issues like this one ? Apologies in advance if there's something obvious that I lazily missed.

I was thinking of the following:

* A programmer's model/spec/description of the CL protocol (I see a bunch of threads on the interwebs but wasn't sure what was current or relevant).

* A description of a typical setup for examining relevant bus traffic. I get the impression that USB HID is the generally accepted abstraction for comms with such devices. I wondered whether folks had typical setups for examining such comms ?

For example - I think I read somewhere that one way of learning more about the CL protocol would be to use Corsair's Windows specific tooling running in a Windows VM on Virtualbox which is set up to enable pass through comms from the Windows VM to the USB controller hardware. Then use Wireshark on the Linux host. I would be very interested in knowing if that's the way to go to learn more.

Maybe such information could be maintained on the OCL project wiki ? I would be happy to help with that if it's the way to go.

Thanks!

Legion2, 2020-01-01:
@raw-bin have a look at https://github.com/Legion2/CorsairLightingProtocol this is an open source  implementation of the current protocol on the device side.

audiohacked, 2020-01-01:
@raw-bin There isn't any knowledge repository for the corsair hardware. Most of the work has been done by using Wireshark and the Windows programs to sniff the USB.

raw-bin, 2020-01-10:
Thanks both! @audiohacked : Do you perchance have any kind of wireshark resources that you've built  up like filter scripts etc ? Or other such goodness that can help n00bs get cracking ? If not - for the problem that I'm seeing what would you suggest I look at first ? All fairly woolly questions I admit but I'm hoping that I can learn from experienced folks and contribute back. Thanks!