hyperkineticnerd/OpenCorsairLink #221: H150i Pro Temperature Mode Not Changing Color

daviscyl, 2019-12-30:
My H150i Pro does not change color under the temperature mode in Ubuntu.

**Command Used:**
sudo ./OpenCorsairLink.elf --debug --device 0 --led channel=0,mode=5,colors=00B2FF:FFFF00:FF0000,temps=30:50:70

**Response:**
LED Channel = 0
LED Mode = 5
Found Color 0: 00B2FF
Found Color 1: FFFF00
Found Color 2: FF0000
Found Color 0: 30
Found Color 1: 50
Found Color 2: 70
Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (1d6b:0002)...
Checking USB device 2 (1d6b:0003)...
Checking USB device 3 (1d6b:0002)...
Checking USB device 4 (1d6b:0002)...
Checking USB device 5 (1d6b:0003)...
Checking USB device 6 (1d6b:0003)...
Checking USB device 7 (1d6b:0002)...
Checking USB device 8 (1d6b:0003)...
Checking USB device 9 (1d6b:0002)...
Checking USB device 10 (1d6b:0003)...
Checking USB device 11 (1d6b:0002)...
Checking USB device 12 (1d6b:0003)...
Checking USB device 13 (1b1c:0c12)...
Corsair product detected. Checking if device is H150i Pro... Dev=0, CorsairLink Device Found: H150i Pro!
Checking USB device 14 (0b05:1867)...
Checking USB device 15 (046d:c52b)...
Checking USB device 16 (413c:2112)...
Checking USB device 17 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: selected device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H150i Pro
AA 12 34 01 00 04 00
Firmware: 1.0.4.0
hardware version returned: AB 12 34 02 00 00 00
A9 12 34 23 01
Temperature 0: 35.10 C
00
41 12 34 00 00 00
Fan 0:  Mode 0x00
        Current/Max Speed 0/0 RPM
00
41 12 34 01 00 00
Fan 1:  Mode 0x00
        Current/Max Speed 0/0 RPM
00
41 12 34 02 00 00
Fan 2:  Mode 0x00
        Current/Max Speed 0/0 RPM
function:corsairlink_asetekpro_pump_mode_read file: protocol/asetekpro/pump.c
pump mode response = 33 12 34 02
pump speed response = 31 12 34 0B 22
Pump:   Mode 0x02 (AsetekProPumpPerformance)
        Current/Max Speed 2850/0 RPM
Setting LED Flag found
Setting LED to TEMPERATURE
5F 12 34 00 00 00
5E 12 34
DEBUG: deinit done

The response seems all fine to me, but the LED just stays at the first color (#00B2FF) no matter how hot the CPU runs.

audiohacked, 2019-12-31:
There's several reasons why the LED isn't changing. As a note, OpenCorsairLink doesn't link CPU temperature with LEDs. The hardware only natively supports Liquid Temperature. I assume you are reading the CPU Temp with another program. As the Liquid Temp is 35 Celsius, and if I remember correctly, that would keep the LED on #00B2FF. I don't think you'll see a difference until the Liquid Temp approaches 50 Celsius.

wishie, 2020-05-10:
Ahh, that makes sense.. I assume this is the same with the H100i Pro RGB that I have just installed. So, if the hardware is measuring 'liquid temperature' then there would be a significant delay in the liquid heating up..

Is there a way in the future to support OpenCorsairLink to read temp from an external program? I guess that would also mean OpenCorsairLink would have to be always running, though..