hyperkineticnerd/OpenCorsairLink #222: --help suggests --machine gives output in machine readable format, but prints nothing

mmattice, 2020-01-06:
**Describe the bug**
--help suggests --machine gives output in machine readable format, but prints nothing

**To Reproduce**
```
mattice:~/proj/mike/corsair/OpenCorsairLink$ sudo  ./OpenCorsairLink.elf  --device 0
Dev=0, CorsairLink Device Found: HX1000i!

Vendor: CORSAIR
Product: HX1000i
Firmware: NA
Temperature 0: 41.20 C
Temperature 1: 30.19 C
Powered: 57719904 (668d.  1h)
Uptime: 3161904 (36d.  14h)
Supply Voltage: 115.00 V
Total Watts: 268.00 W
Output 12v:
        Voltage 12.16 V
        Amps 17.50 A
        Watts 208.00 W
Output 5v:
        Voltage  5.09 V
        Amps 10.75 A
        Watts 55.00 W
Output 3.3v:
        Voltage  3.31 V
        Amps  0.75 A
        Watts  2.00 W
mattice:~/proj/mike/corsair/OpenCorsairLink$ sudo  ./OpenCorsairLink.elf  --device 0  --machine
mattice:~/proj/mike/corsair/OpenCorsairLink$

```

**Expected behavior**
Some sort of machine readable output, json, csv, etc

**Desktop (please complete the following information):**
 - OS: debian buster

audiohacked, 2020-01-07:
`--machine` hasn't been fully implemented in all code bases. Currently only available in one(??) driver.

mmattice, 2020-01-07:
Which driver is that for reference?

mmattice, 2020-01-07:
NM.  Found it.  I hacked in some capabilities for the two devices I've got.  It probably isn't a huge stretch to add the json stuff to everything.

https://github.com/mmattice/OpenCorsairLink/tree/commanderpro_json

The commits to main.c cause it to loop through everything if it's in JSON mode, so it dumps it all in a mostly nicely formatted object.  I'm just throwing that directly at my MQTT server so I can process things and alert from node-red.  Like so:

```
* * * * * root /usr/local/sbin/OpenCorsairLink --json 2>/dev/null | /usr/bin/mosquitto_pub -h mqtt -t hosts/server/corsair -s
```

closed 2020-05-29 by audiohacked