hyperkineticnerd/OpenCorsairLink #223: H80i has an incorrect pump_index

zeroepoch, 2020-01-17:
The `pump_index` for the H80i (no suffix) is incorrect.  It's actually been incorrect almost from the beginning, but it didn't become a problem until 5581c978558d054496114fd9ad02496995d607b1 was committed which started to use `pump_index` in ways it didn't before.  The change is pretty straightforward so I didn't think a pull request would be necessary. Here's the change:

```
diff --git a/device.c b/device.c
index c3ff1ea..6c45f10 100644
--- a/device.c
+++ b/device.c
@@ -115,7 +115,7 @@ struct corsair_device_info corsairlink_devices[] = {
         .lowlevel = &corsairlink_lowlevel_coolit,
         .led_control_count = 1,
         .fan_control_count = 4,
-        .pump_index = 5,
+        .pump_index = 0,
     },
     {
         .vendor_id = 0x1b1c,
```

audiohacked, 2020-05-29:
@zeroepoch How do you know that the index is wrong?

zeroepoch, 2020-05-29:
It didn't work until I changed the index and it also worked before the commit mentioned. The referenced commit changed the index (when it fixed the bug) so seems logical it's the wrong index since it was 0 before and it worked and it's 5 now and it doesn't work and changing it back to 0 works. I hope that explanation makes sense. I wouldn't have known it worked at all if it wasn't for the fact that I used this program years ago when I bought the device and it worked then. That's what prompted me to go bisecting to figure out when it broke.

audiohacked, 2020-06-04:
@zeroepoch Do you have the fan(s) connected to the H80i or to the motherboard?

zeroepoch, 2020-06-05:
Yes I have two fans attached to one connector. The second connector is unused.

zeroepoch, 2020-06-05:
Without the patch I get:
```
$ sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H80i!

Vendor: Corsair
Product: H80i
Firmware: 1.0.7
Temperature 0: 34.04 C
Fan 0:	Quiet Mode (4PIN)
	Current/Max Speed 1309/2511 RPM
Fan 1:	Quiet Mode (4PIN)
	Current/Max Speed 1328/2504 RPM
Fan 2:	Not connected or failed
	Current/Max Speed 0/0 RPM
Fan 3:	Not connected or failed
	Current/Max Speed 0/0 RPM
Pump:	Mode 0x08
	Current/Max Speed 0/0 RPM
```

With the patch I get:
```
$ sudo ./OpenCorsairLink.elf --device 0
Dev=0, CorsairLink Device Found: H80i!

Vendor: Corsair
Product: H80i
Firmware: 1.0.7
Temperature 0: 34.03 C
Fan 0:	Quiet Mode (4PIN)
	Current/Max Speed 1309/2511 RPM
Fan 1:	Quiet Mode (4PIN)
	Current/Max Speed 1328/2504 RPM
Fan 2:	Not connected or failed
	Current/Max Speed 0/0 RPM
Fan 3:	Not connected or failed
	Current/Max Speed 0/0 RPM
Pump:	Mode 0x89
	Current/Max Speed 1309/2511 RPM
```

audiohacked, 2020-06-05:
@zeroepoch If you didn't notice, the RPM for the Pump is the exact same as Fan 0. What your patch is doing is telling the software to treat Fan 0 data as the Pump data. I expect the Pump Speed to be closer to 1000 RPM higher than the Fan Speed.

zeroepoch, 2020-06-06:
@audiohacked that's interesting! I didn't notice that. I guess it's always been like that then because that's the same behavior I get from using a much earlier commit. I see now why you suspected that 0 was also not right.

zeroepoch, 2023-02-23:
Since this issue is quite old now, and I don't know the proper solution, and haven't been using this tool recently, I'm going to close out this old issue.

closed 2023-02-23 by zeroepoch