hyperkineticnerd/OpenCorsairLink #225: Unsupported Generic LED Function with Lighting Node PRO + Suuport for it?

kalinda99, 2020-01-24:
I don't know if this counts as a bug or not since the hardware I'm using isn't on the supported list but OpenCorsairLink does detect my Lighting Node Pro, maybe because it's similar to the Commander Pro? The only difference between them, as far as I know, is that the Lighting can't control fan RPM, it's just for the pretty RGB.

Anyway, I'd love to see support added for the Lighting Node Pro if there is none. I'd be happy to do testing or get whatever info would be needed for it.

**Describe the bug**
When I execute the command sudo OpenCorsairLink --dev=1 --led channel=2,mode=4,colors=00FF00, I get Unsupported Generic LED Function at the end. Here's the full output:

`Dev=0, CorsairLink Device Found: H100i Pro!
Dev=1, CorsairLink Device Found: Lighting Node PRO!

Vendor: Corsair
Product: Lighting Node PRO
Firmware: V0.0.0
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Temperature 3:  0.00 C
Output 12v:  0.00 V
Output 5v:  0.00 V
Output 3.3v:  0.00 V
Fan 0:  Mode: Auto/Disconnected
        PWM: 0%
        RPM: 0
Fan 1:  Mode: Unknown
        PWM: 0%
        RPM: 0
Fan 2:  Mode: Unknown
        PWM: 0%
        RPM: 0
Fan 3:  Mode: Unknown
        PWM: 0%
        RPM: 0
Fan 4:  Mode: Unknown
        PWM: 0%
        RPM: 0
Fan 5:  Mode: Unknown
        PWM: 0%
        RPM: 0
Unsupported Generic LED Function`

It detects all six of my fans, though it doesn't talk to them, or maybe it's just detecting that the device supports six fans.

**Expected behavior**
To control my lights just like I can with my H100i. Maybe I'm doing it wrong? I tried several different modes and combos and I just get that error.

**Desktop:**
 - OS: Arch Linux with kernel 5.4.13
 - Version: I'm using the testing branch package from Arch's AUR, which I just rebuild today so I have the most recent commits.

audiohacked, 2020-01-29:
LED protocol for the Lighting Node Pro (and the Commander Pro) isn't fully mapped yet.

aleczk, 2020-02-15:
I was going to ask the same question, I guess... The error message is a bit different.
I'm a newbie in Linux and I made a mistake of buying 6 freaking LL fans which have RGB with it even tho I keep them black (no color) on windows.

```
aleczk@Aleczk:~/Downloads/OpenCorsairLink-testing$ sudo ./OpenCorsairLink.elf
Dev=0, CorsairLink Device Found: HX750i!
Dev=1, CorsairLink Device Found: Commander PRO!

aleczk@Aleczk:~/Downloads/OpenCorsairLink-testing$ sudo ./OpenCorsairLink.elf --device 1 --led channel=0, mode=2, colors=00FF00
Dev=0, CorsairLink Device Found: HX750i!
Dev=1, CorsairLink Device Found: Commander PRO!

Vendor: Corsair
Product: Commander PRO
Firmware: V0.9.212
Temperature 0:  0.00 C
Temperature 1:  0.00 C
Temperature 2:  0.00 C
Temperature 3:  0.00 C
Output 12v: 11.88 V
Output 5v:  4.98 V
Output 3.3v:  3.33 V
Fan 0:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
Fan 1:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 1429
Fan 2:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 1339
Fan 3:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 1490
Fan 4:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 1496
Fan 5:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 1490
Unsupported LED Function
```

BeardedDevCo, 2020-04-20:
Di you manage to get anywhere with this @audiohacked ? Running elementaryOS 5.1 am in the testing branch but the LED side of the commander pro doesnt seem to be getting recognised

sudo ./OpenCorsairLink.elf --dev=0 --led channel=2,mode=0,colors=00FF00
Dev=0, CorsairLink Device Found: Commander PRO!

Vendor: Corsair
Product: Commander PRO
Firmware: V0.9.212
Temperature 0: 30.01 C
Temperature 1: 40.59 C
Temperature 2: 30.13 C
Temperature 3:  0.00 C
Output 12v: 12.14 V
Output 5v:  4.92 V
Output 3.3v:  0.66 V
Fan 0:	Mode: Auto/Disconnected
	PWM: 50%
	RPM: 0
Fan 1:	Mode: Auto/Disconnected
	PWM: 50%
	RPM: 0
Fan 2:	Mode: Auto/Disconnected
	PWM: 50%
	RPM: 0
Fan 3:	Mode: Auto/Disconnected
	PWM: 50%
	RPM: 0
Fan 4:	Mode: Auto/Disconnected
	PWM: 50%
	RPM: 0
Fan 5:	Mode: Auto/Disconnected
	PWM: 50%
	RPM: 0
Unsupported LED Function

Solarer, 2020-06-14:
I can confirm the observation of @kalinda99 and @BeardedDevCo. I have the Lightning Node Pro connected to my Commander Pro and the output is the same. Since development is moving to liquidctl it is likely that we will get the feature there.
If I have some spare time I might have a look at it. For now you can also try to use OpenRGB for the LEDs and OCL for fan speed.