hyperkineticnerd/OpenCorsairLink #228: H100i GT V2

jkvargas, 2020-02-08:
➜  ~ sudo OpenCorsairLink.elf --device 0 --pump mode=5
Hey guys, I am sorry for opening this bug again, but this is happening using the most updated code from master branch. When trying to set the pump mode to performance it says to me that this is not supported.
I know there was a fix in the past for this, but it seems that it is still broken.
Or maybe I am doing something wrong.

Machine:
Intel(R) Core(TM) i7-5960X CPU @ 3.00GHz
64GB ram
NVIDIA GTX 1080
H100i GT V2

Terminal output:
Dev=0, CorsairLink Device Found: H100i GT V2!

Vendor: Corsair
Product: H100i GT V2
Firmware: 2.8.0.0
Temperature 0: 29.30 C
Temperature 1: 29.30 C
Temperature 2: 29.30 C
Fan 0:	Mode 0xEB
	Current/Max Speed 0/0 RPM
Unsupported Pump Function
Pump:	Mode 0x05
	Current/Max Speed 2040/0 RPM
Setting pump to mode: 5
Unsupported Pump Function

audiohacked, 2020-02-09:
Don't use `master` branch.

droopymiller, 2020-03-27:
Had this same issue and using the `testing` branch solved my issue.  Also thank you for dedicating your time to this piece of software.  I really appreciate what you have done!

closed 2020-05-29 by audiohacked