hyperkineticnerd/OpenCorsairLink #229: h100i not detected

ChameleonScales, 2020-02-15:
Hi,
Thanks for making this tool.
I'm trying to get it working but it seems my H100I is not detected.
I run:
`sudo ./OpenCorsairLink.elf --debug`
and I get:
```
Checking USB device 0 (1058:10a8)...
Checking USB device 1 (1d6b:0003)...
Checking USB device 2 (1532:0042)...
Checking USB device 3 (13fd:1840)...
Checking USB device 4 (1d6b:0002)...
Checking USB device 5 (1d6b:0003)...
Checking USB device 6 (046d:c31c)...
Checking USB device 7 (13fd:1840)...
Checking USB device 8 (05e3:0608)...
Checking USB device 9 (8087:0029)...
Checking USB device 10 (1d6b:0002)...
Checking USB device 11 (1d6b:0003)...
Checking USB device 12 (1b1c:0c18)...
Checking USB device 13 (05e3:0608)...
Checking USB device 14 (048d:8297)...
Checking USB device 15 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: selected device_number = -1
```
then I run
`sudo ./OpenCorsairLink.elf --device 0`
and get
`Detected 0 device(s), submitted device 0 is out of range`

**Desktop:**
 - OS: Ubuntu 19.10
 - Cooler: Corsair H100I Platinum
 - Motherboard: GIGABYTE X570 AORUS MASTER

 - OpenCorsairLink version: v0.9.0.0-46db

picklenebula, 2020-02-17:
Same Issue for me

```
sudo ./OpenCorsairLink.elf --debug 

Checking USB device 0 (1d6b:0003)...
Checking USB device 1 (1852:7022)...
Checking USB device 2 (046d:082d)...
Checking USB device 3 (0d8c:0166)...
Checking USB device 4 (1d6b:0002)...
Checking USB device 5 (1d6b:0003)...
Checking USB device 6 (046d:c333)...
Checking USB device 7 (046d:c085)...
Checking USB device 8 (0a5c:21e8)...
Checking USB device 9 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: selected device_number = -1
```

```
sudo ./OpenCorsairLink.elf --device 0 

Detected 0 device(s), submitted device 0 is out of range
```

OS: Manjaro 18.0
Cooler: Corsair H100i v2 
Motherboard: Gigabyte AB350-Gaming 3
OpenCorsairLink version: shamefully, I am a git noob, it is the master branch as of today's date.

ChameleonScales, 2020-02-18:
The USB cable is connected in my case, so this issue is still unsolved for me.

picklenebula, 2020-02-18:
> The USB cable is connected in my case, so this issue is still unsolved for me.

Out of curiosity is the cooler detected by the system?
Say with `lsusb` or something similar?

jonasmalacofilho, 2020-02-21:
@ChameleonScales the H100i Platinum is not yet supported (see: #146).

ChameleonScales, 2020-02-22:
out of lsusb, here are some I get :
Bus 001 Device 004: ID 1b1c:0c18 Corsair ITE Device(8595)
Bus 001 Device 002: ID 048d:8297 Integrated Technology Express, Inc. ITE Device(8595)
is that it?

closed 2020-05-29 by audiohacked