hyperkineticnerd/OpenCorsairLink #231: H100i RGB PRO XT

andyNewman42, 2020-04-14:
H100i RGB Pro XT is not detected or supported.  This is different from the RGB Pro, as it sounds like it uses coolit pump.

The device ID for this is 0c20

```
sudo lsusb -d 1b1c:0c20 -vvv

Bus 003 Device 004: ID 1b1c:0c20 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c20 
  bcdDevice            1.00
  iManufacturer           1 Corsair Inc.
  iProduct                2 H100iRGBPROXT
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

```

I would like to contribute to this development,
Is there anything I can do that can brick/break the AIO? If I start testing and sending requests to this device, what should I look out for in terms of breaking things?

andyNewman42, 2020-04-14:
Note: 

This is different from the h100i pro, and is not a duplicate request, as the XT version is a new version

audiohacked, 2020-04-16:
@andyNewman42 
Wonderful!

There is a chance for bricking, but I'm mostly sure that to get access to the AIO bootloader is quite involved. The only other issue is "committing" an AIO profile to the hardware, again, something that has to be investigated.

andyNewman42, 2020-04-16:
@audiohacked Do you have any documentation for adding a new profile?

Willburn, 2020-05-10:
I too have an H100I RGB PRO xt just curious if this is being worked on as I would love to adjust fan speeds :) (Home server and want it to go more quiet)