hyperkineticnerd/OpenCorsairLink #241: ld: multiple definition of 'scanlist' error

Abh15h3k, 2020-05-16:
OS: Artix Linux ( using openrc )

Im trying install opencorsairlink-git from the AUR using yay. but i get the following error. but it installs properly on a Arch system.

`cc main.o device.o logic/options.o logic/options_fan.o logic/options_led.o logic/options_pump.o logic/print.o logic/scan.o logic/settings/commanderpro.o logic/settings/hydro_asetek.o logic/settings/hydro_asetekpro.o logic/settings/hydro_coolit.o logic/settings/psu.o drivers/asetek.o drivers/asetekpro.o drivers/commanderpro.o drivers/coolit_old.o drivers/coolit.o drivers/dongle.o drivers/rmi.o drivers/unsupported.o protocol/asetek/core.o protocol/asetek/fan.o protocol/asetek/led.o protocol/asetek/pump.o protocol/asetek/temperature.o protocol/asetekpro/core.o protocol/asetekpro/fan.o protocol/asetekpro/led.o protocol/asetekpro/pump.o protocol/asetekpro/temperature.o protocol/commanderpro/core.o protocol/commanderpro/fan.o protocol/commanderpro/power.o protocol/commanderpro/temperature.o protocol/rmi/core.o protocol/rmi/power.o protocol/rmi/temperature.o protocol/rmi/time.o protocol/coolit/core.o protocol/coolit/fan.o protocol/coolit/led.o protocol/coolit/pump.o protocol/coolit/temperature.o lowlevel/asetek.o lowlevel/commanderpro.o lowlevel/coolit.o lowlevel/rmi.o -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -std=c99 -Iinclude -I/usr/include/libusb-1.0  -DVERSION=\"v0.9.0.0-e2bd\" -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now -lm -lusb-1.0  -o OpenCorsairLink.elf
/usr/sbin/ld: logic/scan.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: multiple definition of `scanlist'; main.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: first defined here
/usr/sbin/ld: logic/settings/commanderpro.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: multiple definition of `scanlist'; main.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: first defined here
/usr/sbin/ld: logic/settings/hydro_asetek.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: multiple definition of `scanlist'; main.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: first defined here
/usr/sbin/ld: logic/settings/hydro_asetekpro.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: multiple definition of `scanlist'; main.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: first defined here
/usr/sbin/ld: logic/settings/hydro_coolit.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: multiple definition of `scanlist'; main.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: first defined here
/usr/sbin/ld: logic/settings/psu.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: multiple definition of `scanlist'; main.o:/tmp/yaourt-tmp-abhishek/aur-opencorsairlink-git/src/OpenCorsairLink/include/logic/scan.h:30: first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:150: OpenCorsairLink.elf] Error 1`

Abh15h3k, 2020-05-16:
Well this was a buzzkill. but after downgrading to gcc 9.3.0 the packages builds without error

Abh15h3k, 2020-05-17:
Ok. So i have a workaround for people who might not want to downgrade their gcc to 9.3.0.

The linker errors out because scanlist ( from include/logic/scan.h ) is getting redefined in main.c and logic/scan.c

so i removed the "scanlist[10]" from logic/scan.h and declared them individually in main.c and scan.c (with different variable name, otherwise the linker will still report one redefinition error).

so in include/logic/scan.h:
struct corsair_device_scan
{
    struct corsair_device_info* device;

    /** libusb device structures */
    struct libusb_device_handle* handle;
} ~~scanlist[10]~~ ;

so in main.c right after all the includes i added:
struct corsair_scan_device scanlist[10];

and in logic/scan.c: ( right after the includes again )
struct corsair_scan_device slist[10];

I'm sure this is not best/correct way to do it. but this is what i could figure what in the small time i spent tinkering with this. 
Hopefully this info helps fixing (the right way)  the linker errors

duxet, 2020-05-22:
It's easier to just add `CFLAGS += -fcommon` line to `Makefile`

Abh15h3k, 2020-05-22:
> It's easier to just add `CFLAGS += -fcommon` line to `Makefile`
Oh cool. 
But does that exactly do?

audiohacked, 2020-05-25:
Thanks for submitting a bug report. The fix should be in the testing branch.

P.S. I will warn you that I do not support AUR/PKGBUILDs.

closed 2020-05-25 by audiohacked