hyperkineticnerd/OpenCorsairLink #242: Device Support: iCUE H115i RGB Pro XT

blake-newman, 2020-05-26:
The **iCUE H115i RGB Pro XT** is not detected as a supported device. This AOI is relatively new 2020 release. 

**Device details:**

```
lsusb -d 1b1c:0c21 -vvv

Bus 001 Device 010: ID 1b1c:0c21 Corsair 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c21 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               7

```

blake-newman, 2020-05-26:
This article seems to suggest that the pump is CoolIT. 

On my initial testing i could get the device to be recognised

```
{
        .vendor_id = 0x1b1c,
        .product_id = 0x0c21,
        .device_id = 0x00,
        .name = "H115i RGB PRO XT",
        .read_endpoint = 0x01 | LIBUSB_ENDPOINT_IN,
        .write_endpoint = 0x00 | LIBUSB_ENDPOINT_OUT,
        .driver = &corsairlink_driver_coolit,
        .lowlevel = &corsairlink_lowlevel_coolit,
        // .led_control_count = .,
        // .fan_control_count = .,
        // .temperature_control_count = .,
        // .pump_index = .,
    },
```

```
Corsair product detected. Checking if device is H115i RGB PRO XT... 
---- Packet dump: -----------------------------
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
-----------------------------------------------
Dev=0, CorsairLink Device Found: H115i RGB PRO XT!
Checking USB device 11 (1b1c:1c02)...
Checking USB device 12 (05e3:0610)...
Checking USB device 13 (0b05:18f3)...
Checking USB device 14 (1d6b:0002)...

DEBUG: scan done, start routines
DEBUG: selected device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H115i RGB PRO XT
Firmware: 1.6.17
Speed: 00 00
Max Speed: 00 00
Fan 0:  Not connected or failed
        Current/Max Speed 0/0 RPM
Speed: 00 00
Max Speed: 00 00
```

I'm unsure what to configure the properties to or how to discover the values they should be set to.

blake-newman, 2020-05-26:
I see you have already begun some work on this @audiohacked 

https://github.com/jonasmalacofilho/liquidctl/issues/80#issuecomment-597967627

Is there anything I can do to support?

clmpo, 2020-06-03:
I also have the H115i RGB Pro XT.

@audiohacked do you have suggestions or hints on how we could help? I have the impression that the commands for this new CoolIt protocol are different from the one already implemented.

audiohacked, 2020-06-05:
So I had some time to figure out how to read the AIO status and it looks like its very similar to the H115i Platinum. Setup Wireshark under Windows, and Start there.

clmpo, 2020-06-06:
Good news, indeed the Platinum H115i profile seems to work for the XT
    {
        .vendor_id = 0x1b1c,
        .product_id = 0x0c21,
        .device_id = 0xFF,
        .name = "H115i RGB PRO XT",
        .read_endpoint = 0x01 | LIBUSB_ENDPOINT_IN,
        .write_endpoint = 0x01 | LIBUSB_ENDPOINT_OUT,
        .driver = &corsairlink_driver_platinum,
        .lowlevel = &corsairlink_lowlevel_platinum,
        .led_control_count = 3,
        .fan_control_count = 2,
        .pump_index = 0,
    },

Corsair product detected. Checking if device is H115i RGB PRO XT... Dev=0, CorsairLink Device Found: H115i RGB PRO XT!

DEBUG: scan done, start routines
DEBUG: selected device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H115i RGB PRO XT
Firmware: 1.1.22
Temperature 0: 31.81 C
Speed: 28 02
Fan 0:  Unknown (STUB)
        Current/Max Speed 552/65535 RPM
Speed: 28 02
Fan 1:  Unknown (STUB)
        Current/Max Speed 552/65535 RPM
Speed: 8B 07
Max Speed: 14 1F
Pump:   Mode 0x00
        Current/Max Speed 1931/65535 RPM
DEBUG: deinit done

The problem is that after the first run of the program the fans speed goes to 1807 RPM... and a reboot is necessary for bringing back the quiet mode. Any idea why?

Subsequent runs:

sudo ./OpenCorsairLink.elf --debug --dump --device 0
Corsair product detected. Checking if device is H115i RGB PRO XT... Dev=0, CorsairLink Device Found: H115i RGB PRO XT!

DEBUG: scan done, start routines
DEBUG: selected device_number = 0
DEBUG: shortcuts set
DEBUG: init done
Vendor: Corsair
Product: H115i RGB PRO XT
Firmware: 1.1.22
Temperature 0: 28.05 C
Speed: 0F 07
Fan 0:  Unknown (STUB)
        Current/Max Speed 1807/65535 RPM
Speed: 0F 07
Fan 1:  Unknown (STUB)
        Current/Max Speed 1807/65535 RPM
Speed: 8B 07
Max Speed: 00 1C
Pump:   Mode 0x00
        Current/Max Speed 1931/65535 RPM
DEBUG: deinit done

I also managed to get some captures of the USB interaction between the AIO and iCUE... What would be useful besides reading the temperature and speed is to be able to 
1. be able to set the fans RPM
2. be able to set the pump RPM
3. disable the RGB (the most useless feature in the entire universe :) )
4. set profiles (quiet, balanced, etc) like iCUE

How would you go about reverse engineering the protocol from the captures? Any online materials you recommend? I have a good knowledge of wireshark for network stuff but I never looked at USB or HID.