hyperkineticnerd/OpenCorsairLink #243: Future of OpenCorsairLink

audiohacked, 2020-06-07:
On June 6th 2020, I decided to retire OpenCorsairLink as explained in the `README.md`. I also set the Project to Read-Only/Archived. This morning I realized I would like to collect/merge any and all branches for posterity and as a resource. I'd like to Archive this project by January 1st 2021 and move future development to [liquidctl](https://github.com/jonasmalacofilho/liquidctl). I believe python is fast/mature enough and liquidctl has a great framework for supporting Corsair products on Linux.

There are several changes and improvements I would like to make to OpenCorsairLink, but they might become breaking and full API changes.

I'd really appreciate some thoughts and comments on the project.

clmpo, 2020-06-07:
First of all, let me thank you for working on this project and spending valuable time and energy developing this. This project has been quite useful as an example of what Corsair protocols look like and as a starting point for hacking on my own device. 

I think that there is a place for both liquidctl and OpenCorsairLink. In my opinion, a resource lightweight (memory and cpu wise) tool that is easily embedded in any other language is a valuable thing.

In your opinion, what is the amount of time needed for reverse engineering the different protocols vs actually implementing support?

Should we maybe, as a community, get a little bit more vocal asking from Corsair to open their protocol? As both consumers and power-users, I think we should try to nudge the company for more support, especially opening up their protocol which would benefit both liquidctl and other projects like OpenCorsairLink

Thanks again!

audiohacked, 2020-06-07:
Using Windows and Wireshark, I can sort of figure out a protocol and almost fully map status, fan, and LED packets after about 3 days. And at least for me, I can see the patterns in the Wireshark dumps, but I also need to have direct interaction with iCUE and the hardware.

The hardest part is finding motivation to do it.

wispoffates, 2020-06-09:
I wanted to say thanks for all the hard work you've put it.  I just started my adventure into AIO coolers and picked a Corsair product specifically because this this tool existed.

 Sadly while this tool can read the values of the 115i platinum, liquidctl cannot yet (though some work looks to have been started).

Have fun on whatever it is you do next!

audiohacked, 2020-06-09:
@wispoffates I have a branch that should allow liquidctl to read reliably the 115i platinum.

Solarer, 2020-06-13:
Sad news that this project is being closed. Thank you a lot for your work, controlling the fan speed works well but have not figured out the LEDs yet. 
Will all features of OpenCorsairLink be migrated into liquidctl in the end or will the features of the tools not overlap?

audiohacked, 2020-06-13:
@Solarer I plan on migrating as much of OpenCorsairLink as I can into liquidctl

jonasmalacofilho, 2020-06-13:
Those who want to help port drivers to liquidctl may find [liquidctl#129 (comment)](https://github.com/jonasmalacofilho/liquidctl/issues/129#issuecomment-640258429) to be helpful as a first guide.  Please also feel free to open and issue and ask questions.

---

@audiohacked and all others that have contributed to OpenCorsairLink, thank you!

Not only you have helped the open-source community in general, but you were also responsible for me having faith that, if I needed, I would be able to write the tools and drivers for the devices I wanted to get.

_(Which turned out to be the case, and that eventually led to liquidctl)._

zararashigal, 2025-02-07:
Yeah but what if i don't have a liquid cooler but want to change my fan curve... <_<

zararashigal, 2025-02-07:
I don't think electronics go well with water. Liquid cooling was a mistake.