hyperkineticnerd/OpenCorsairLink #246: [Obsidian 1000D Commander Pro] Setting fan modes have no effect.

max0x7ba, 2020-08-27:
**Describe the bug**
Setting fan modes with Obsidian 1000D Commander Pro have no effect.

**To Reproduce**
Let's view the fan status:
```
$ sudo /usr/local/bin/OpenCorsairLink.elf --device 1
Dev=0, CorsairLink Device Found: H110i!
Dev=1, CorsairLink Device Found: Obsidian 1000D Commander Pro!

Vendor: Corsair
Product: Obsidian 1000D Commander Pro
Firmware: V0.8.210
Temperature 0: 35.93 C
Temperature 1: 27.97 C
Temperature 2: 34.93 C
Temperature 3:  0.00 C
Output 12v: 11.92 V
Output 5v:  4.98 V
Output 3.3v:  3.36 V
Fan 0:	Mode: Auto/Disconnected
	PWM: 100%
	RPM: 1374
Fan 1:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 750
Fan 2:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 750
Fan 3:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 752
Fan 4:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 751
Fan 5:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
```

There are 5 fans connected to this fan controller and it displays RPM for each connected fan correctly. 

But status "Mode: Auto/Disconnected" doesn't seem to be accurate at all.

Let's set fan modes:
```
$ sudo /usr/local/bin/OpenCorsairLink.elf --device 1 --fan channel=0,mode=5 > /dev/null
$ sudo /usr/local/bin/OpenCorsairLink.elf --device 1 --fan channel=1,mode=5 > /dev/null
$ sudo /usr/local/bin/OpenCorsairLink.elf --device 1 --fan channel=2,mode=5 > /dev/null
$ sudo /usr/local/bin/OpenCorsairLink.elf --device 1 --fan channel=3,mode=5 > /dev/null
$ sudo /usr/local/bin/OpenCorsairLink.elf --device 1 --fan channel=4,mode=5 > /dev/null
$ sudo /usr/local/bin/OpenCorsairLink.elf --device 1 
Dev=0, CorsairLink Device Found: H110i!
Dev=1, CorsairLink Device Found: Obsidian 1000D Commander Pro!

Vendor: Corsair
Product: Obsidian 1000D Commander Pro
Firmware: V0.8.210
Temperature 0: 36.03 C
Temperature 1: 27.72 C
Temperature 2: 35.11 C
Temperature 3:  0.00 C
Output 12v: 11.91 V
Output 5v:  4.98 V
Output 3.3v:  3.36 V
Fan 0:	Mode: Auto/Disconnected
	PWM: 100%
	RPM: 1361
Fan 1:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 750
Fan 2:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 750
Fan 3:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 752
Fan 4:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 751
Fan 5:	Mode: Auto/Disconnected
	PWM: 0%
	RPM: 0
```

It seems that setting fan modes with Obsidian 1000D Commander Pro have no effect at all.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.5 LTS
 - Version
```
$ git log -n1
commit 61d336a61b85705a5e128762430dc136460b110e (HEAD -> testing, origin/testing, origin/HEAD)
Author: Audiohacked <audiohacked@gmail.com>
Date:   Sat Jun 6 08:48:41 2020 -0600

    add a backup of the old README before setting the project to READ-ONLY

```

**Additional context**
Setting fan modes for H110i (device 0) works correctly.

Raphty, 2020-11-24:
I have the same issue
on Elementary OS 5.1 (Based on Ubuntu 18.04 LTS)

max0x7ba, 2021-06-11:
https://www.kernel.org/doc/html/latest/hwmon/corsair-cpro.html kernel module now supports Obsidian 1000D Commander Pro out of the box:

```
$ sensors
corsaircpro-hid-3-4
Adapter: HID adapter
in0:          11.94 V  
in1:           5.00 V  
in2:           3.37 V  
fan1 4pin:    897 RPM
fan2 4pin:    900 RPM
fan3 4pin:    900 RPM
fan4 4pin:    828 RPM
fan5 4pin:    857 RPM
temp1:        +42.4°C  
temp2:        +30.2°C  
temp3:        +46.4°C  
...
```

closed 2021-06-11 by max0x7ba